fairecasoimeme/ZiGate #189: Cannot change the Extended PANID

pipiche38, 2019-04-13:
With ZGUI (as well as with the API), I'm not able to change the Extended PANID

Here after is the screenshot of:
- setting Channel to 26
- Setting ExtPANID to 0x00124b00198d7521
- Setting COORDINATOR

-> Start NWK
-> Network State

All Commands get an Ok Status !!
[image]

pipiche38, 2019-04-13:
However if you set 0x0 as the extended PANID parameter it is taken into account and you get the Extended PANID reset!

pipiche38, 2019-04-15:
After further investigation, I have been able to force a specific Extended PANID only after an Erase PDM.

So for me there are some inconsistency : 

(1) If you need Erase PDM prior to set a specific Extended PANID (and that could make sense), if we are not in such case, Zigate should return an error and not send 0x8000 with 0x00 Status
(2) In case you set ExtendedPANID to 0 whatever is the situation of the PDM (Erased or not), you get a new Extended PANID. Which then is inconsistent with (1)

pipiche38, 2019-04-25:
any idea ?

ISO-B, 2019-05-04:
If PANID is 0x00 when you start forming network ZiGate will generate random id.

You probably can't and should not be able to change PANID after forming and starting network. This is probably reason why you can't change panid without PDM reset. More info: JN-UG-3113 v1.5 Chapter 5.1.1

There might be bug in function ZPS_eAplAibSetApsUseExtendedPanId() that is used to set panid, which lets you to reset panid at anytime. If thats case we can't do nothing, because its inside nxp library files that can't be edited because they don't share source code for those.

pipiche38, 2019-05-04:
Would it makes sense to have a deterministic way implemented in ZiGate ?

And si return an error if network is already formed ?

ISO-B, 2019-05-04:
Yes, that might be good solution to prevent changing coordinators panid after network formation. I will implement it later this weekend.

PR #230 (2019-08-23): Prevent changing Ext PANID after stack is started
Changing Extended PANID in coordinator mode wont do anything after stack
is started. For this reason we now prevent changing it. Sending command
in this case will result status E_SL_MSG_STATUS_STACK_ALREADY_STARTED
0x04. Fixes #189
commit a415ff00 (2019-08-23): Prevent changing Ext PANID after stack is started

Changing Extended PANID in coordinator mode wont do anything after stack
is started. For this reason we now prevent changing it. Sending command
in this case will result status E_SL_MSG_STATUS_STACK_ALREADY_STARTED
0x04. Fixes #189
closed 2019-08-21 by pipiche38