fairecasoimeme/ZiGate #190: APS Data Failure - 0x8702 - Problem with non-existing device

pipiche38, 2019-04-16:
When sending a Command ( ON or OFF ) to a device which is not in the Zigbee network anymore, I'm getting:

0xd4 - Unicast frame does not have a route available but it is buffered for automatic resend

However, has this device is not present on the network, I 'm expecting a more precise failure at the end of the 'buffered for automatic resend' !

How to reproduce the problem:

-1 Pair un Ikea Tradfri Bulb with the Zigate
-Power Off (electrical/physical) the Ikea Tradfri Bulb ==> Not on Zigbee network anymore
- Send a ON CMD

You'll receive : Status: d4 - Unicast frame does not have a route available but it is buffered for automatic resend

I'm expecting after a certain TimeOut a more formal error informing that the transaction has failed , as the 0xd4 is not a failure as such

pipiche38, 2019-04-16:
Funny enough, it looks like for others devices, I'm getting a different Status.

For example for a Xiaomi Plug or an other Ikea Tradfri Bulb

```
 Status: e9 - No acknowledgement received when expected
```

pipiche38, 2019-04-25:
It looks that even with a functioning Ika Tradrfri Plug we are getting such APS Failure 0xe9 code !!!!!!

pipiche38, 2019-04-30:
This is becoming a very annoying issue.
Even we are receiving 0xe9 message for devices which have not send anything !

ISO-B, 2019-05-04:
> -Power Off (electrical/physical) the Ikea Tradfri Bulb ==> Not on Zigbee network anymore
> You'll receive : Status: d4 - Unicast frame does not have a route available but it is buffered for automatic resend
>

It is still in network, because it has not unpaired it is just powered off. If you reset device without zigate knowing it still thinks that device is in its network untill it is unpaired/removed. If it would not be in network table any more you would need to repair it to zigate after turning power on. So for me this seems to be correct status.

> It looks that even with a functioning Ika Tradrfri Plug we are getting such APS Failure 0xe9 code !!!!!!
>

You are sending command to plug and it does what is should do, but you still get status 0xe9?

> Even we are receiving 0xe9 message for devices which have not send anything !
>

One reason for receiving 0xe9 without sending anything that some kind of  keep alive type message is sent without you knowing from zigate and it won't get response.

pipiche38, 2019-05-04:
@ISO-B thank, that is indeed the conclusion I came too.
Would it be possible from the Zigate stack to bring additional information with the APS Failure code to inform if the code is related to a Zigbee stack (like the 0xe9) or if this is in regards to User command ?

ISO-B, 2019-05-04:
I will need to look in to it. How I could easiest reproduce this behavior?

pipiche38, 2019-05-29:
@ISO-B don't know if you have been able to duplicate, but I found a workaround on my side. 
So I have no problem to close this issue if you want

ISO-B, 2019-05-30:
No I didn't find way to easily reproduce this. If problem comes back I can try again

closed 2019-05-30 by ISO-B