fairecasoimeme/ZiGate #192: Problem with status of lights

nerdosity, 2019-04-27:
Hi, I am using home assistant with doudz plugin.
Everything pairs good (more or less), but IKEA bulbs are driving me mad.
Once in a while, one of them disappear, then if I reset it that light won't work good anymore.

More details here:

https://github.com/doudz/homeassistant-zigate/issues/43

nerdosity, 2019-04-27:
This is happening with one of my two outlets (bought 30 minutes ago).

`
2019-04-27 19:13:17 DEBUG (Thread-13) [zigate] Waiting for status message for command 0x0092
2019-04-27 19:13:17 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x98\x02\x10\x8f\x02\x10\x92\x02\x10\x03'
2019-04-27 19:13:17 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-04-27 19:13:17 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'008f0092'
2019-04-27 19:13:17 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:143, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:17 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-04-27 19:13:17 DEBUG (Thread-13) [zigate] STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:0, sequence:143, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:19 DEBUG (Thread-259) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
2019-04-27 19:13:19 DEBUG (Thread-259) [custom_components.zigate] Update attribute for device LUMI lumi.sensor_motion.aq2 (4688) 00158d0001fa419d {'endpoint': 1, 'value': False, 'type': <class 'bool'>, 'attribute': 0, 'name': 'presence', 'addr': '4688', 'data': False, 'cluster': 1030, 'expire': 10}
2019-04-27 19:13:19 DEBUG (Thread-5) [custom_components.zigate.binary_sensor] Event received: {'endpoint': 1, 'value': False, 'attribute': 0, 'device_type': 'lumi.sensor_motion.aq2', 'type': 'bool', 'entity_id': 'zigate.00158d0001fa419d', 'name': 'presence', 'ieee': '00158d0001fa419d', 'addr': '4688', 'data': False, 'cluster': 1030, 'expire': 10}
2019-04-27 19:13:39 DEBUG (Thread-15) [zigate] REQUEST : 0x0092 b'\x02YQ\x01\x01\x01'
2019-04-27 19:13:39 DEBUG (Thread-15) [zigate] Msg to send b'009200069f025951010101'
2019-04-27 19:13:39 DEBUG (Thread-15) [zigate] Encoded Msg to send b'01021092021002169f0212595102110211021103'
2019-04-27 19:13:39 DEBUG (Thread-15) [zigate] Waiting for status message for command 0x0092
2019-04-27 19:13:39 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x87\x02\x10\x90\x02\x10\x92\x02\x10\x03'
2019-04-27 19:13:39 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-04-27 19:13:39 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00900092'
2019-04-27 19:13:39 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:144, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:39 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-04-27 19:13:39 DEBUG (Thread-15) [zigate] STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:0, sequence:144, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:43 DEBUG (Thread-12) [zigate] REQUEST : 0x0092 b'\x02YQ\x01\x01\x01'
2019-04-27 19:13:43 DEBUG (Thread-12) [zigate] Msg to send b'009200069f025951010101'
2019-04-27 19:13:43 DEBUG (Thread-12) [zigate] Encoded Msg to send b'01021092021002169f0212595102110211021103'
2019-04-27 19:13:43 DEBUG (Thread-12) [zigate] Waiting for status message for command 0x0092
2019-04-27 19:13:43 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x86\x02\x10\x91\x02\x10\x92\x02\x10\x03'
2019-04-27 19:13:43 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-04-27 19:13:43 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00910092'
2019-04-27 19:13:43 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:145, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:43 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-04-27 19:13:43 DEBUG (Thread-12) [zigate] STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:0, sequence:145, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:44 DEBUG (Thread-20) [zigate] REQUEST : 0x0092 b'\x02YQ\x01\x01\x00'
2019-04-27 19:13:44 DEBUG (Thread-20) [zigate] Msg to send b'009200069e025951010100'
2019-04-27 19:13:44 DEBUG (Thread-20) [zigate] Encoded Msg to send b'01021092021002169e0212595102110211021003'
2019-04-27 19:13:44 DEBUG (Thread-20) [zigate] Waiting for status message for command 0x0092
2019-04-27 19:13:44 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x85\x02\x10\x92\x02\x10\x92\x02\x10\x03'
2019-04-27 19:13:44 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-04-27 19:13:44 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00920092'
2019-04-27 19:13:44 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:146, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:44 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-04-27 19:13:44 DEBUG (Thread-20) [zigate] STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:0, sequence:146, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:51 DEBUG (Thread-4) [zigate] REQUEST : 0x0092 b'\x02YQ\x01\x01\x01'
2019-04-27 19:13:51 DEBUG (Thread-4) [zigate] Msg to send b'009200069f025951010101'
2019-04-27 19:13:51 DEBUG (Thread-4) [zigate] Encoded Msg to send b'01021092021002169f0212595102110211021103'
2019-04-27 19:13:51 DEBUG (Thread-4) [zigate] Waiting for status message for command 0x0092
[... cut ...]
2019-04-27 19:13:55 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00950092'
2019-04-27 19:13:55 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:149, packet_type:146, error:b'', lqi:0
2019-04-27 19:13:55 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-04-27 19:13:55 DEBUG (Thread-6) [zigate] STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:0, sequence:149, packet_type:146, error:b'', lqi:0
`

pipiche38, 2019-05-06:
This looks more a HA/Plugin issue rather than a Zigate/Firmware one.

gotakk, 2019-05-22:
J'ai le même genre de problème avec ma philips hue
https://github.com/doudz/homeassistant-zigate/issues/46

closed 2020-11-23 by fairecasoimeme