fairecasoimeme/ZiGate #194: Wrong decoding of INNR Remote ( 0x8085 doesn't report the reallity )

pipiche38, 2019-05-03:
It seems that there is some descrapancy in the way the Zigate is decoding and providing information to the 0x8085 message with the INNR RC110 remote controller

the Left/Right button are always providing the same infos, while when sniffing the messages with Ubiqua we see the right things

With Ubiqua, we see
Level Control: Move Up , Move Down, Stop

with Zigate, we see only Release Down, Stop

pipiche38, 2019-05-03:
[image]

pipiche38, 2019-05-03:
[image]

fairecasoimeme, 2019-05-04:
Normally fixed in v3.1a

pipiche38, 2019-07-08:
Thanks , it works !

closed 2019-07-08 by pipiche38