fairecasoimeme/ZiGate #196: Legrand Netatmo wireless switch support

Xavier-M, 2019-05-28:
Hello,

I would like to know if the Legrand Netatmo Wireless Switch is now supported? Or if there is a chance it will be supported soon?

I can see it was not working on the blog post: https://faire-ca-soi-meme.fr/domotique/2018/09/24/test-legrand-celiane-by-netatmo-zigate/

I need to buy around 5-6 wireless switchs and it would be really nice if it is Legrand Celiane like the others.

Thanks

fairecasoimeme, 2019-06-03:
Hello,
Sorry but for the moment I don't retest this device. 
Fred

Xavier-M, 2019-06-23:
Hello,

I just bought one and in fact it is working!

lboue, 2019-07-03:
Hello Xavier,

Could you tell us how did you make it work.
Datasheet: [Legrand - 067723](http://docdif.fr.grpleg.com/general/MEDIAGRP/NP-FT-GT/F02491FR-00.pdf)

Xavier-M, 2019-07-07:
Hello,

In fact I did nothing special. It joins the network directly, then I pressed on/off a lot and did some discover devices (I'm using Home Assistant and doudz's plugin: https://github.com/doudz/homeassistant-zigate).

And then I have a new entity with On/Off (reading your document it seems it could also be used as a variator, but I don't think it works). I got this after the discover:
```
{
            "addr": "46f1",
            "discovery": "auto-discovered",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 4,
                                    "data": " Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
                                    "name": "manufacturer",
                                    "value": "Legrand"
                                },
                                {
                                    "attribute": 5,
                                    "data": " Remote switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
                                    "name": "type",
                                    "type": "str",
                                    "value": "Remote switch"
                                },
                                {
                                    "attribute": 65533
                                },
                                {
                                    "attribute": 0,
                                    "name": "zcl_version",
                                    "value": null
                                },
                                {
                                    "attribute": 1,
                                    "name": "application_version",
                                    "value": null
                                },
                                {
                                    "attribute": 2,
                                    "name": "stack_version",
                                    "value": null
                                },
                                {
                                    "attribute": 3,
                                    "name": "hardware_version",
                                    "value": null
                                },
                                {
                                    "attribute": 6,
                                    "name": "datecode",
                                    "value": null
                                },
                                {
                                    "attribute": 7,
                                    "name": "power_source",
                                    "value": null
                                },
                                {
                                    "attribute": 61440
                                },
                                {
                                    "attribute": 8
                                },
                                {
                                    "attribute": 9
                                },
                                {
                                    "attribute": 10
                                },
                                {
                                    "attribute": 11
                                },
                                {
                                    "attribute": 16384
                                }
                            ],
                            "cluster": 0
                        },
                        {
                            "attributes": [
                                {
                                    "attribute": 32,
                                    "name": "battery_voltage",
                                    "type": "float",
                                    "value": 0.0
                                },
                                {
                                    "attribute": 65533
                                }
                            ],
                            "cluster": 1
                        },
                        {
                            "attributes": [
                                {
                                    "attribute": 85,
                                    "data": false,
                                    "name": "present_value",
                                    "type": "bool",
                                    "value": false
                                }
                            ],
                            "cluster": 15
                        }
                    ],
                    "device": 260,
                    "endpoint": 1,
                    "in_clusters": [
                        0,
                        3,
                        15,
                        32,
                        1,
                        64513
                    ],
                    "out_clusters": [
                        3,
                        6,
                        8,
                        0,
                        64513,
                        25
                    ],
                    "profile": 260
                }
            ],
            "generictype": "",
            "info": {
                "addr": "46f1",
                "ieee": "XXX",
                "last_seen": "2019-07-07 20:16:22",
                "lqi": 147,
                "mac_capability": "10000000"
            }
        }
```

Xavier-M, 2019-07-07:
In fact I have 2 switchs, both are working, but I can see that the second one has more attributes:
```
 {
                            "attributes": [
                                {
                                    "attribute": 81,
                                    "name": "out_of_service",
                                    "type": "bool",
                                    "value": false
                                },
                                {
                                    "attribute": 85,
                                    "data": false,
                                    "name": "present_value",
                                    "type": "bool",
                                    "value": false
                                },
                                {
                                    "attribute": 111,
                                    "name": "status_flags",
                                    "value": null
                                },
                                {
                                    "attribute": 65533
                                }
                            ],
                            "cluster": 15
                        },
```

pipiche38, 2019-07-08:
My understanding is that Legrand devices are shipped out with a basic firmware, and you need to use their gateway to upgrade first the device firmware and get all features . 
So Legrand is and remain not open and I wouldn't consider using them

Xavier-M, 2019-07-08:
Maybe, I don't really know. I have 8 micromodules and 2 wireless switches and it works without having the legrand gateway. But I don't know if I would have more features with it and upgrades...

pipiche38, 2019-10-09:
Wireless switch works with the Zigate !

pipiche38, 2019-10-09:
Can be closed

Xavier-M, 2019-10-09:
What do you use for the Level Up / Level Down?

pipiche38, 2019-10-09:
Basically on the Domoticz Zigate plugin we have a switch selector which is mapped to the Legrand Wireless switch.

On --> On
Off --> Off
Long press On --> Up
Long press Off -> Down
release (after Long Press) --> Release

pipiche38, 2019-10-09:
The thing which is not supported for now on the Wireless is the Led management!

Xavier-M, 2019-10-09:
Ok thank you and the "long press" is a specific attribute or do you just use "present_value"?

I'm using HASS with Doudz's plugin and it's the only attribute that changes when I use the wireless switch.

pipiche38, 2019-10-09:
@Xavier-M , nobody's perfect ;-) Joking

it comes from message 0x8095 and 0x8085 (this was introduced in 3.0f, but corrected with 3.1f and not the standard  Individual Attribute response

* 0x8085 for long press and release
* 0x8095 for the On/Off. 
   For instance, the first time you do On you'll get the 0006 cluster with 0000 attribute, but if you press it once more, you won't and you'll get a 0x8095 instead

lboue, 2019-10-09:
Did someone tried the new double switch version?

[Connected 2-gang ON/OFF wireless control for lighting and sockets / Ref 067724](http://docdif.fr.grpleg.com/general/MEDIAGRP/NP-FT-GT/F02504EN-00.pdf
)

pipiche38, 2019-10-09:
Yes, that is the one I did the test with.
So far only the Led is not managed

pipiche38, 2020-06-25:
This shoudl be close.Legrand devices are supported and can be connected to Zigate.

closed 2020-06-25 by fairecasoimeme