fairecasoimeme/ZiGate #197: Aqara 86 sw2 Pairing to zigate after pairing to Xiaomi Gateway

deennoo, 2019-05-29:
Bonjour,

Zigate USB fw 3.0e
Plugin Zigate Domoticz 4.3.0
Channel 15 en mode CE

J'ai un Inter Xiaomi Aqara 86SW2 (inter double sans fils de première génération).
Il a toujours été liée a ma gateway Xiaomi.
Aujourd'hui je souhaite l'inclure a la Zigate, et l'appairage ne fonctionne pas.

Ma demarche : 
Depuis l'app mihome gateway j'efface le device. j'attend 5 minutes, puis debranche la gateway (il m'est arriver pour d'autre device qu'elle les reappaires meme sans etre en inclusion.

Je passe la zigate en mode appairage
Je reset le device : 
 - long press pendant 10 secondes : 4 clignotements de la led en bleu
 - le device announce ce fait (3 clignotement rapide)
 -  5 secondes plus tard 3 clignotements lent qui indique que l appareil n'est dans aucun reseau zigbee

J'ai essayer avec ZGUI : idem, pas d'appairage possible.

Peux tu essayer de ton coté ?

fairecasoimeme, 2019-05-30:
OK,
J'ai normalement le matériel. J'essaie ASAP
Fred.

fairecasoimeme, 2019-06-03:
Salut,
J'ai eu le temps de bien pousser les tests. J'ai le même environnement que toi (hormis ta version de ZiGate qui, à mon avis est le problème...)

Comme tous les Xiaomi, celui-ci ne déroge pas à la règle. Pas d'OTA request, donc pas de demande de mise à jour. (J'ai laissé tourner toute la nuit avec un sniffer)

Je pense que le problème vient de ta version de ZiGate. la 3.0e n'est pas stable et engendre un problème d'appairage au bout de quelques devices.

Je te conseille donc de mettre à jour la ZiGate en 3.0f et de reprendre les tests.

Fred.

deennoo, 2019-06-14:
ok merci des tests.

Il apparaît ce soir que c'est un problème de Zigate Pleine, 60 devices actif et 10 fantômes encore en mémoire

closed 2019-06-14 by deennoo