fairecasoimeme/ZiGate #198: Message 0x804E should also have Source Address

pipiche38, 2019-06-02:
When you send a message 0x004E, you expect to have a 0x804E, however there is no source address in the 0x804E.

Would be great to have such information as for the other messages !

fairecasoimeme, 2019-06-03:
Hi,
I don't really understand by source address because, you normally get each source address from neighbor like this (for example) :

Type: 0x804E (Mgmt LQI Response)
  SQN: 0x68
  Status: 0x00
  Nb Table Entries: 2
  Start Index: 0
  Nb Table List Count: 2
  Neighbor 0:
    Extended Pan ID: 0xEE81BB144F75C53C
    Extended Address: 0x158D0001C5D10F
    Nwk Address: 0x0000
    LQI: 255
    Depth: 0
    Flags: 0x48
    Device Type: Coordinator
    Permit Joining: Unknown
    Relationship: Parent
    RxOnWhenIdle: Yes
  Neighbor 1:
    Extended Pan ID: 0xEE81BB144F75C53C
    Extended Address: 0xB57FFFE3686DA
    Nwk Address: 0x09A3
    LQI: 255
    Depth: 2
    Flags: 0x12
    Device Type: End Device
    Permit Joining: Off
    Relationship: Child
    RxOnWhenIdle: No

Could you precise ?
Fred

pipiche38, 2019-06-03:
By Source address I'm talking about from which router this List of neighbours is sent !
The Nwk Address that are listed here are the neighbours of this "Source Address"

fairecasoimeme, 2019-06-25:
source address added to v3.1a

pipiche38, 2019-06-25:
Excellent, will make the things much reliable

pipiche38, 2019-07-02:
Merci

doudz, 2019-07-18:
Doc not up to date

fairecasoimeme, 2019-07-18:
For me, It's up to date. What informations is missing ?

pipiche38, 2020-09-30:
Fixed 3.1a

closed 2020-09-30 by pipiche38