fairecasoimeme/ZiGate #199: Xiaomi sensors non-functional after x hours

Fracz1111, 2019-06-06:
Hello all !
First, thank you for this project i am liking it so much !

Here is my issue :
After around 30 hours of working (after a jeedom reboot), my devices (mostly Xiaomi) are not functionning anymore.

I am joining screenshots with the example of my door sensor.
What I have noticed :
- Zigate OK : no log error
- Door sensor up and sync (as the others)
- Historization activated
- No more activation after several hours : for example on the screenshot, when I left the home the sensor has not noticed my left

:(

Any ideas guys ?

Thank you very much for your help !

> Zigate Infos :
> Version plugin
> 1.2.3
> Version lib
> 0.28.4
> Version firmware
> 3.0f 

[image]
[image]
[image]

Fracz1111, 2019-06-12:
I paste here the log at debug level from this morning > in this case today : no actions detected

```
[2019-06-12 04:50:30][DEBUG][zigate](Thread-505) : File already existing, make a backup before
[2019-06-12 04:55:30][DEBUG][zigate](Thread-506) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 04:55:30][DEBUG][zigate](Thread-506) : Saving persistent file
[2019-06-12 04:55:30][DEBUG][zigate](Thread-506) : File already existing, make a backup before
[2019-06-12 05:00:30][DEBUG][zigate](Thread-507) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:00:30][DEBUG][zigate](Thread-507) : Saving persistent file
[2019-06-12 05:00:30][DEBUG][zigate](Thread-507) : File already existing, make a backup before
[2019-06-12 05:05:30][DEBUG][zigate](Thread-508) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:05:30][DEBUG][zigate](Thread-508) : Saving persistent file
[2019-06-12 05:05:30][DEBUG][zigate](Thread-508) : File already existing, make a backup before
[2019-06-12 05:10:30][DEBUG][zigate](Thread-509) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:10:30][DEBUG][zigate](Thread-509) : Saving persistent file
[2019-06-12 05:10:30][DEBUG][zigate](Thread-509) : File already existing, make a backup before
[2019-06-12 05:15:30][DEBUG][zigate](Thread-510) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:15:30][DEBUG][zigate](Thread-510) : Saving persistent file
[2019-06-12 05:15:30][DEBUG][zigate](Thread-510) : File already existing, make a backup before
[2019-06-12 05:20:31][DEBUG][zigate](Thread-511) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:20:31][DEBUG][zigate](Thread-511) : Saving persistent file
[2019-06-12 05:20:31][DEBUG][zigate](Thread-511) : File already existing, make a backup before
[2019-06-12 05:25:31][DEBUG][zigate](Thread-512) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:25:31][DEBUG][zigate](Thread-512) : Saving persistent file
[2019-06-12 05:25:31][DEBUG][zigate](Thread-512) : File already existing, make a backup before
[2019-06-12 05:30:31][DEBUG][zigate](Thread-513) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:30:31][DEBUG][zigate](Thread-513) : Saving persistent file
[2019-06-12 05:30:31][DEBUG][zigate](Thread-513) : File already existing, make a backup before
[2019-06-12 05:35:31][DEBUG][zigate](Thread-514) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:35:31][DEBUG][zigate](Thread-514) : Saving persistent file
[2019-06-12 05:35:31][DEBUG][zigate](Thread-514) : File already existing, make a backup before
[2019-06-12 05:40:31][DEBUG][zigate](Thread-515) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:40:31][DEBUG][zigate](Thread-515) : Saving persistent file
[2019-06-12 05:40:31][DEBUG][zigate](Thread-515) : File already existing, make a backup before
[2019-06-12 05:45:31][DEBUG][zigate](Thread-516) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:45:31][DEBUG][zigate](Thread-516) : Saving persistent file
[2019-06-12 05:45:31][DEBUG][zigate](Thread-516) : File already existing, make a backup before
[2019-06-12 05:50:31][DEBUG][zigate](Thread-517) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:50:31][DEBUG][zigate](Thread-517) : Saving persistent file
[2019-06-12 05:50:31][DEBUG][zigate](Thread-517) : File already existing, make a backup before
[2019-06-12 05:55:31][DEBUG][zigate](Thread-518) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 05:55:31][DEBUG][zigate](Thread-518) : Saving persistent file
[2019-06-12 05:55:31][DEBUG][zigate](Thread-518) : File already existing, make a backup before
[2019-06-12 06:00:31][DEBUG][zigate](Thread-519) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 06:00:31][DEBUG][zigate](Thread-519) : Saving persistent file
[2019-06-12 06:00:31][DEBUG][zigate](Thread-519) : File already existing, make a backup before
[2019-06-12 06:05:31][DEBUG][zigate](Thread-520) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 06:05:31][DEBUG][zigate](Thread-520) : Saving persistent file
[2019-06-12 06:05:31][DEBUG][zigate](Thread-520) : File already existing, make a backup before
[2019-06-12 06:10:31][DEBUG][zigate](Thread-521) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 06:10:31][DEBUG][zigate](Thread-521) : Saving persistent file
[2019-06-12 06:10:31][DEBUG][zigate](Thread-521) : File already existing, make a backup before
[2019-06-12 06:15:32][DEBUG][zigate](Thread-522) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-12 06:15:32][DEBUG][zigate](Thread-522) : Saving persistent file
[2019-06-12 06:15:32][DEBUG][zigate](Thread-522) : File already existing, make a backup before
[... cut ...]
[2019-06-12 10:11:36][DEBUG] : result callZiGate Array (     [result] =>       [success] => 1 )
[2019-06-12 10:11:37][DEBUG] : callZiGate get_last_responses
[2019-06-12 08:11:37][DEBUG][root](Thread-5) : Message received in socket
[2019-06-12 08:11:37][DEBUG][root](Thread-5) : {'action': 'get_last_responses', 'args': ''}
[2019-06-12 08:11:37][DEBUG][root](Thread-5) : {'result': '
', 'success': True}
[2019-06-12 10:11:37][DEBUG] : result callZiGate Array (     [result] =>       [success] => 1 )
```

Fracz1111, 2019-06-13:
After looking deeper in the logs, this case is attached to the bug #81 
Problème stabilité #81
I will check 3 things and comment the results asap :
- avoid my usb extender (10 cm) but we don't know
- change SD card to SSD disk
- reinstall and resync all my devices

Here are the logs before "crash" , hope it will help somebody
```
[2019-06-13 15:21:05][DEBUG] : Array (     [device] => Array         (             [info] => Array                 (                     [bit_field] => 0100000000000010                     [manufacturer_code] => 1037                     [ieee] => 00158d0002478376                     [descriptor_capability] => 00000000                     [last_seen] => 2019-06-13 13:21:05                     [server_mask] => 0                     [power_type] => 0                     [lqi] => 105                     [mac_capability] => 10000000                     [id] => 5                     [addr] => bdd4                 )              [addr] => bdd4             [endpoints] => Array                 (                     [0] => Array                         (                             [endpoint] => 1                             [device] => 24321                             [in_clusters] => Array                                 (                                     [0] => 0                                     [1] => 3                                     [2] => 65535                                     [3] => 6                                 )                              [profile] => 260                             [clusters] => Array                                 (                                     [0] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [attribute] => 1                                                             [data] => 3                                                             [value] => 3                                                             [name] => application_version                                                         )                                                      [1] => Array                                                         (                                                             [attribute] => 4                                                             [data] => LUMI                                                             [value] => LUMI                                                             [name] => manufacturer                                                         )                                                      [2] => Array                                                         (                                                             [attribute] => 5                                                             [name] => type                                                             [type] => str                                                             [data] => lumi.sensor_magnet.aq2                                                             [value] => lumi.sensor_magnet.aq2                                                         )                                                      [3] => Array                                                         (                                                             [attribute] => 65281                                                             [data] => 0121d10b0328150421a84305213500062401000000000a210000641000                                                             [unit] => V                                                             [value] => 3.025                                                             [type] => float                                                             [name] => battery_voltage                                                         )                                                  )                                              [cluster] => 0                                         )                                      [1] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [attribute] => 0                                                             [name] => onoff                                                             [type] => bool                                                             [data] =>                                                              [value] =>                                                          )                                                  )                                              [cluster] => 6                                         )                                  )                              [out_clusters] => Array                                 (                                     [0] => 0                                     [1] => 4                                     [2] => 65535                                 )                          )               [... cut ...]
[2019-06-13 13:21:05][DEBUG][requests.packages.urllib3.connectionpool](Thread-1) : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=klQwqHfkkjkjysuWuxiCei76d6O1Sbbj HTTP/1.1" 200 16
[2019-06-13 13:21:05][DEBUG][root](Thread-1) : Jeedom reply :  {'success': True}
[2019-06-13 13:23:48][DEBUG][zigate](Thread-704) : Auto saving /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2019-06-13 13:23:48][DEBUG][zigate](Thread-704) : Saving persistent file
[2019-06-13 13:23:48][DEBUG][zigate](Thread-704) : File already existing, make a backup before
```

pipiche38, 2019-06-13:
I would have a look to your case from 2 angles:

(1) which is related to the Network and Link quality between the sensors and Zigate
(2) Zigate Plugin

I cannot really comment on the (2) as I have no knowledge on it

As regards to (1), it is important to understand that the ZigBee network rely on Mesh network. So having only one sensors (powered by battery) is not ideal.
So you might have a look to the Interference level. for instance to you have any overlap between Wifi 2.4 network channels and Zigbee channels (here is an interesting compilation of informations): https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Co-existence-of-IEEE-802.15.4-at-2.4-GHz-and-Zigbee.md

You might have a look on the level of Mesh (if of course) you have zigbee routers (most-often all main powered devices are)

Fracz1111, 2019-06-14:
Hello Pipiche,
Thank you for your help and the very intersting link.
it could be indeed, my only question is why always after 24 hours I have this result...
So I go on that angle :
> in order to isolate the root cause, I first switch off the 2.4GHz wifi on my box (I let the 5Ghz on for the moment).
In addition, I will monitor a sensor located at 5 meters from the zigate dongle.
No mesh for me because I don't have any other powered zigbee devices.
Concerning the module, I will focus on it once the other points will be validated.

I will have the answer tomorrow, I will update this thread with the results.
thank you

doudz, 2019-06-15:
Looks similar to https://github.com/doudz/homeassistant-zigate/issues/34
@mvdkleijn reports the same problem.
The common point is that both plugin (Jeedom et Home assistant) are based on my python zigate lib

@Fracz1111 what's the LQI of the sensor ?

Fracz1111, 2019-06-18:
Thanks for your help, LQI 146 for the sensor i'm monitoring.
Exactly it sounds similar, sorry for adding a same issue.
[image]
thank you again

Fracz1111, 2019-06-30:
Hello
My tests :
- without usb extender : same results
- changing sd card : same results
> only services restant fix temporaly the problem.
> i conclude something wrong on library zigate
> try to workaround with cron every 8 jour to restart services automatically
Doudz, if you have any perspective of fix, dont hésitante to share
thank you again !

doudz, 2019-06-30:
strange but could the same as
https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues/188
or
https://community.home-assistant.io/t/another-component-for-zigate/58417/378

doudz, 2019-06-30:
@Fracz1111 could you share the last "Raw packet received" line from your log ?

Smanar, 2019-06-30:
For your first issue (this one with the osram plug) it's normal, there is a new find bug in Xiaomi device. Xiaomis devices can't work if you have an osram plug as router, a problem in their firmware (Xiaomi side).
I haven't memorised the link, but you can find more information on deconz and zigbee2mqtt github.

Fracz1111, 2019-07-09:
> 
> 
> strange but could the same as
> [Jeedom-Zigate/jeedom-plugin-zigate#188](https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues/188)
> or
> https://community.home-assistant.io/t/another-component-for-zigate/58417/378

Thank you for that,
I will experience independently : 

- a cooling solution in order to test
 Not very sure about that supposition : I get this problem since a while, and also when the weather was really really bad ...

- a crontab to restart the service regularly 
 if you have any any about the trigger to use i'm aware (for now I will automate a periodic restart I think)

@doudz > 
> @Fracz1111 could you share the last "Raw packet received" line from your log ?
I'm sorry but I don't know how to do this and my search on google had not helped me, could you detail your request, sorry for that and thank you again for your help

Have a nice day

nouknouk, 2019-08-03:
Same issue here. 

Hardware setup:
- an USB-TTL zigate, first version (command #748 on 2017-11-18).
- a 'DIY home automation box' based on an Atom Z8350 mini computer
- a self powered USB hub (DLink DUB-H7) with several devices connected on it: Zigate, Zwave sitck, Arduino, french 'Téléinfo' module.
- an USB cable of 3 meters between USB hub and zigate
- zigate is located in the basement of my house, where temperature is quite constant

included in the network:
- 4 xiaomi aquara [switch keys](https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcQD94lVRNmkuYkiXIgOahTyupr4utLHdBz7sMgPf7P4ONJJPYGc)
- 5 xiaomi [temperature & humidity sensors (2nd gen)](https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcRaIolWP9wJEtvrehjdEtcuxJbGC8eQl7ernLMt-XYkxbUBhCXWzw)

Software setup:
- a Debian 9..7 is installed on the mini PC
- one udev rule to recognize the Zigate and make a constant symlink pointing to zigate's port (/dev/zigate)
- my own Zigate driver, based on NodeJS and available on github: https://github.com/nouknouk/node-zigate 

What I experienced so far:

- the zigate was working well for months in this configuration ; no freeze, or any other issue experienced (including no range issue, etc...).

- Initially, I had the firmware 3.0a ; a 1 month ago I upgraded to firmware 3.0f, then a week ago to firmware 3.1a. Each time I did a complete EEPROM clear, and restarted from a white page, new network, re-included each device, etc...)

- The issues started raising around 1 month ago.

- On my side, I get the zigate working from 4 to 24 hours, and then no more update is received from the temp sensors. No more notification when I click on Xiaomi switchs.

- Once the problem arised, there is no change to get it working back.

- Once the problem arised, I can send a "GET VERSION" command to the zigate, but i never get any form of response.

- If I close the serial connexion and reconnect immediatly after, the issue disappears and the zigate is up and running ... for the next 4-24 hours.

- I just tried to leave the zigate in the same conditions except I completely removed the plastic box around it, to ensure there is no heating issue.

nouknouk, 2019-08-04:
Feedback from attempt made yesterday:

- removed the plastic case of the zigate, to eliminate the heating hypothesis.
- connexion to zigate port on 2019-08-03 22:28
- implemented a periodic check which sends every 30 seconds a "GET VERSION" command to the zigate and ensure it got the response before next attempt.

Result:

- zigate was no more responding 11 hours after (on 2019-08-04 09:16)

- when the 'GET VERSION' command is sent, no feedback at all (neither status response, not the VERSION response itself) ; whereas 30 seconds earlier the zigate answered normally to the same request
- no frame has been send/received in the meanwhile.

logs:
```
// statup of connexion
[2019-08-03 22:28:31.556] successfully connected to device '/dev/zigate'.

/// everything behaves well until 2019-08-04 09:15:32

// last zigate response:
[2019-08-04 09:15:32.912] sending frame: 01 00 10 00 00 10 03
[2019-08-04 09:15:32.914] sending command:  { type: { id: 16, name: 'version', statusExpected: true, responseExpected: 'version' }, timestamp: 2019-08-04T07:15:32.912Z }
[2019-08-04 09:15:32.922] response received:  { type: { id: 32768, name: 'status', typeHex: '8000' }, status: { id: 0,name: 'success' }, sequence: 0, relatedTo: { id: 16, name: 'version', statusExpected: true, responseExpected: 'version' }, rssi:0 }
[2019-08-04 09:15:32.925] response received:  { type: { id: 32784, name: 'version', typeHex: '8010' }, major: 3, installer: 794, rssi: 0 }

// 30 seconds later, sending again GET_VERSION
[2019-08-04 09:16:02.912] sending frame: 01 00 10 00 00 10 03
[2019-08-04 09:16:02.914] sending command:  { type: { id: 16, name: 'version', statusExpected: true, responseExpected: 'version' }, timestamp: 2019-08-04T07:16:02.912Z }

// 30 seconds later, no response received ; check fails.
[2019-08-04 09:16:32.913] [ERROR] last zigate check failed ; zigate stick is no more responding.
```

nouknouk, 2019-08-04:
another check this morning: 

- serial connexion to zigate started at 11:39

- zigate was no more responding at 12:31, less than one hour after.

- once the zigate failed, a call to (the new) SET_LED command gives the same result as a call to GET_VERSION command: no frame received back (status or anything else).

please find attached the whole debug logs gathered for this attempt, from the connexion to the zigate until the failure.

[20190804-zigate.log](https://github.com/fairecasoimeme/ZiGate/files/3464846/20190804-zigate.log)

note: at the end you'll see I try manually to send commands 'led_set', for which no response is received.

nouknouk, 2019-08-04:
**breaking news:** 

I noticed during the last 2 attempts (and it was probably the same before) that each time I loose the connexion to the zigate, I've got an **error in system logs**, correlated i ntime with the failure of the zigate. `sudo dmesg | grep pl2303` returns rows like:

```
[ 1085.431390] pl2303 ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 2320.313292] pl2303 ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 5714.117383] pl2303 ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 5714.121878] pl2303 ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
```

I found some similar errors on the net, basically: _Those messages mean that the device crashed.  It completely stopped sending USB packets._

Therefore I think it's **either an issue with the USB-TTL hardware, or with its kernel driver**.

my kernel version: `Linux domo 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux`

pipiche38, 2019-08-04:
Could be related to the USB-TTL module.

CC: @fairecasoimeme

doudz, 2019-08-04:
The pl2303 USB module is known to have problem sometimes , that's why the new module is based on cp2102

ruimarinho, 2019-08-04:
Is the new cp2102 compatible with all
zigate versions? I have both (one with the plastic case without screws from 2018 and another one with screws from 2019).

nouknouk, 2019-08-04:
I may have a cp2102 in the bottom of a box ... somewhere, ordered on aliexpress months ago (initially foreseen for usb connexions to esp8266)
 
May I make some tests with it, or the TTL module sold on the zigate.fr shop has specific stuff compared to mine ?

@doudz : do the known problems include the behaviors I mentionned ?
Can issues start some day without priori notice, or it's something that usually happens from the beginning ? because until one month ago, my zigate was working fine.

doudz, 2019-08-04:
@fairecasoimeme can say more, but as far as I know any USB ttl module would work.

Modules sold on zigate.fr have nothing special

ISO-B, 2019-08-05:
@nouknouk nice debugging with this issue. If you find your cp2102 compare it's pinout to pl2303 if pin order is same then it should be plug-n-play. If pin order is not same you can use jump wires to wire it correctly.

Still it is weird that 3.0a firmware worked without problems so long. It might be possible that PL2303 has started breaking recently, but I don't think that this is very likely. Most likely explanation might be that newer ZiGate firmwares transfer more data using TTL which somehow makes adapter to fail. I hope that you will find that cp2102 and could try if it solves this problem.

nouknouk, 2019-08-05:
@ISO-B I think too the cash period is somehow proportional to the amount of messages exchanged: since i put in place my 'auto check' by sending a GET_VERSION every 30 seconds, the crash average period is now around  2 hours, compared tonthe 24 hours previously.

I also agree, there is no suspicion this is related to the firmware ; there may be many other factors (likebthe fact that i had to open the usb stick box and that connexions between the rzdio and ttl modules looked a bit loosely coupled, maybe a kernel update on my debian done in the same days, etc...)

Anyway, I'll try to test with cp2102 by the end of the week and give you some feedback.

Thanks again all for your support.

fairecasoimeme, 2019-08-07:
Hi, 
the CP2102 sold on https://zigate.fr is not special on hardware part. You can buy the same on aliexpress.

The only thing is : 
I program this material with a new product descrip tion (ZiGate). It's easier for developer to attach the module. 
I change too the current out to 250mA (by default It's 150mA) to improve ZiGate power.

Fred

nouknouk, 2019-08-07:
many thanks for your feedback.

I found my cp2102 (exactly the same as the one in the photo of zigate shop) and replaced my pl2303 with this cp2102.

It seems so far there's a huge improvement in stability: with my periodic 30s message sent, crassh used to happen every 2-4 hours. With cp2102, I didn't had any crash during the 24 last hours. Hope it'll continue like that.

> I program this material with a new product descrip tion (ZiGate). It's easier for developer to attach the module.
> I change too the current out to 250mA (by default It's 150mA) to improve ZiGate power.

Didn't know there was kind of updatable 'firmware' in such little piece of hardware. I would be interrested to upgrade both name & curent out limit ; 
- can we do it on our own in a simple way, or there's the need for specific software / hardware ?
- is your custom cp2102 fw open sourced ?

doudz, 2019-08-09:
as workaround, my plugins ask for zigate version every 5min, if no answer I close the connection and reconnect.
it's some kind of NOOP command (since zigate doesn't have one)

fairecasoimeme, 2019-08-09:
@nouknouk you can download this 
https://www.silabs.com/documents/public/software/install_USBXpress_SDK.exe

you can follow this kind of tuto : https://silabs-prod.adobecqms.net/content/usergenerated/asi/cloud/content/siliconlabs/en/community/groups/interface/forum/jcr:content/content/primary/qna/cp210x_communication-yYH2.social.$%7BstartIndex%7D.15.html

With utility you can update product description and power out

Fred

nouknouk, 2019-08-10:
Some feedback: no more crash of the zigate since the replacement of my TTL module 4 days ago.
It looks like the pl2303 TTL module was well the root cause of my troubles.

tekaeren, 2019-08-12:
I had problems with missing aqara sensors all the time.
I've just replaced th pl2303 with cp2102 and ugraded from 3.0f to 3.1a.

I'll also let know if it works correctly now.

Fracz1111, 2019-08-12:
Thank you so much for your help,
I ordered a new ttl module on the zigate shop hoping it will solve the problem,
After the tests, i will post again with the results
thanks

ISO-B, 2019-08-15:
Thank you for @nouknouk for testing things. Let's wait couple more confirms about this issue before closing.

mvdkleijn, 2019-08-16:
I also ordered a new module. I'll try that and report back.

Fracz1111, 2019-09-22:
I do confirm, with the new module ordered on the zigate's store, everything ok since 2 weeks, i also have added some new devices , thanks for the help !

mvdkleijn, 2019-10-09:
Unfortunately, my HA is down and out for now... I therefore cannot confirm from my side yet.

tekaeren, 2019-10-15:
My results after last 3 months with CP2102:
- devices are not disconnecting... when ZiGate is available
- when You restart ZiGate You may loose devices

For me it was the update and restart of HomeAssistant, after first one everything was working fine. After second I've lost over 12 out of 14 connected sensors.

closed 2020-11-23 by fairecasoimeme