fairecasoimeme/ZiGate #200: Problem with 0x804E response

pipiche38, 2019-06-10:
When receiving a 0x804E , I'm getting relationship equal to 0x03 !! what is the meaning ?

While looking after the NXP JN-UG-3101-1.pdf ( ZigBee PRO Stack User Guide ), a value of 3 is not a parent, not a child not a sibling, so what is it ???

Last if I look to the documentation, the u2Relationship is coded on a 3-bits value, while Zigate return only 2 bits !

```
struct {
              unsigned u1Reserved1:1;
              unsigned u2Relationship:3;
              unsigned u2RxOnWhenIdle:2;
              unsigned u2DeviceType:2;
              unsigned u6Reserved2:6;
              unsigned u2PermitJoining:2;
};
```
From the Zigate Firmware
```
u8Bytes  =  sApsZdpEvent.uLists.asNtList[u8Values].uAncAttrs.u2DeviceType;
u8Bytes |=  ( sApsZdpEvent.uLists.asNtList[u8Values].uAncAttrs.u2PermitJoining << 2 );
u8Bytes |=  ( sApsZdpEvent.uLists.asNtList[u8Values].uAncAttrs.u2Relationship << 4 );
u8Bytes |=  ( sApsZdpEvent.uLists.asNtList[u8Values].uAncAttrs.u2RxOnWhenIdle << 6 );
```

ISO-B, 2019-08-22:
From JN-UG-3113 v1.2 Page (397-398)
> u2Relationship:3 is a 3-bit value representing the neighbouring node’s
relationship to the local node:
>- 0: Neighbour is the parent
>- 1: Neighbour is a child
>- 2: Neighbour is a sibling (has same parent)
>- 3: None of the above
>- 4: Neighbour is a former child

I don't know what 3 is. Could it be 3 if node is not neighbor at all. Don't know why it would be listed then.
There is also ZPS_tsNwkActvNtEntry which uses u2Relationship as 2 bit. It is quite weird that NXP has decided to use it as 3 bit in this case and still named it like it is 2 bit. 

I suggest that we will change code so it will return 3 bit value. It will break compatibility with older versions, because we will need to  return u16 instead of u8.

doudz, 2019-08-22:
To me, 3 means the node is physically next to the asked router but is not a child of it and has a different parent.
Imagine the following topology :
[image]
If we ask Router 1 for neighbours, it should reply with
Zigate is a parent : 0
Node 1 is a child : 1
Router 2 is a sibling : 2
Node 2 is not a child and has a different parent : 3

And I know when we should have 4 :roll_eyes: 
Maybe in my example, Node 1 is a former child of ZiGate ? Not a direct child but on the same branch

pipiche38, 2019-08-22:
@doudz interesting

While I can understand 'former child', in your example, this mean that the stack is not doing properly its house-cleaning and we have some remaining nodes in the Neighbourhood table which should be removed.

doudz, 2019-08-22:
I'm not sure.
We are asking for physical neighbours, so (in my example) if node 1 is closed enough to ZiGate, if I ask ZiGate, I could get 4 ? Or am I wrong ?

doudz, 2019-08-22:
OR (I don't know) still in my example, Router 1 is a former child of Zigate ? Because Router 1 is a child of ZiGate but Router 1 has children too
So Router 1 is both a child and a parent so we could get 4 ? ( I never get 4 in my test)

pipiche38, 2019-08-22:
> 
> 
> OR (I don't know) still in my example, Router 1 is a former child of Zigate ? Because Router 1 is a child of ZiGate but Router 1 has children too (parent of node 1)
> So Router 1 is both a child and a parent so we could get 4 ? ( I never get 4 in my test)

probably more realistic

ISO-B, 2019-08-22:
Has anyone managed to receive more than one parent? This can be possible now as ZiGate returns value as 2 bit instead of 3 bit it should return 4 as 0.

pipiche38, 2019-08-22:
> Has anyone managed to receive more than one parent? This can be possible now as ZiGate returns value as 2 bit instead of 3 bit it should return 4 as 0.

Not on my end, but I'm receiving a lot of 'None' 0x03 on a very stable network ( Prod ) which looks a bit strange

pipiche38, 2020-09-30:
Not related to firmware. It is just implementation of the ZIgbee norm

closed 2020-09-30 by pipiche38