fairecasoimeme/ZiGate #201: How to interpret 0x804A - Management Network Update response

pipiche38, 2019-06-11:
When issuing a Management Network Update request with the following arguments

scanDuration: 0x05
scanCount: 0x01
scanMask: all channels

In the response 0x804A, How do you interpret ?

> u8EnergyValuesList is a pointer to the first in the set of reported energy-level measurements (the value 0xFF indicates there is too much interference on the channel
What is exactly energy-level measurements . Does that mean that if I have 0xFE I have a strong energy-level measured and that if good ?

fairecasoimeme, 2019-06-18:
Hi,
I think Energy level is the power of transmission on canal. 0xFF indicate interference on channel and 0xFE is the strong value but in reality you'll never get this value.
Indeed, this value depend on hardware. Apparently you could convert this value to dBm value but I miss a piece of information.

If I find this I'll let you know.

pipiche38, 2019-06-18:
Fred
Thanks a lot for the work.
I think that 0x804A must be enhanced to the the Source Short Address .

pipiche38, 2019-06-19:
@fairecasoimeme , for information here is a report I'm getting

```
 Network Energy Level Report
 -----------------------------------------------
 nwkid     Tx  Failure   11   15   19   20   25   26
  ca62     40        0  171  172  162  161  179  175
  2d25     43        0  175  194  168  165  176  196
  f775     32        0  187  183  164  170  177  187
  9596     36        0  192  196  171  163  175  202
  eb73     30        0  180  184  161  170  180  187
  fc80     24        0  191  185  169  162  178  191
  83c2     23        0  204  210  205  206  190  207
  1f33     33        0  174  188  172  167  179  188
  0fc4     40        0  175  179  172  174  166  177
  aa38     28        0  187  189  180  177  179  191
```

closed 2019-07-02 by pipiche38