fairecasoimeme/ZiGate #202: 0x004A do not trigger a Management Network Update request

pipiche38, 2019-06-17:
Here is the command that I'm sending to ZIgate.
 
004A 0000061888000101000000

I'm expecting in return a Scan for Interferences/Energies on each Zigbee Chanel.
- Scan Duration is set to 1s
- Channel Targets are all Zigbee Channels
- Scan Count is set to 1

I'm expecting in return (or few seconds later ) a 0x804A message, but never come !

fairecasoimeme, 2019-06-18:
I found this (if it can help): 

[image]

I try for the first time this command with ZWGUI and a sniffer. All channel mask value is 0x07fff800 because it seems they're counting from channel 0

The command works but my device is lost after ... my device search a route to 0x0001... It's weird. If I find other things i tell you.

Fred

fairecasoimeme, 2019-06-18:
Ok,

I tested with tradfri router.

I send this command 
00 4A 00 0A 3F 84 F8 02 00 00 00 03 02 00 00

target : 84f8
scan channel : 2000000
Scan duration : 3
Scan count : 2

Here is the result :

80 4A 00 0D E7 11 00 00 0A 00 00 02 00 00 00 01 A4 9C 
80 4A 00 0D E0 11 00 00 02 00 00 02 00 00 00 01 AE 99 

Type: 0x804A (Mgmt Nwk Update Notify)
  SQN: 0x11
  Status: 0x00
  Total Tx: 10
  Tx Failures: 0
  Scanned Channels: 0x02000000
  Scanned Channels List Count: 1
  Value 0:  0xA4
Type: 0x804A (Mgmt Nwk Update Notify)
  SQN: 0x11
  Status: 0x00
  Total Tx: 2
  Tx Failures: 0
  Scanned Channels: 0x02000000
  Scanned Channels List Count: 1
  Value 0:  0xAE

Fred

pipiche38, 2019-06-18:
Does that mean we cannot use 0x0000 (Zigate) as the target address ?

This is something which doesn't sound clear for me:
- Zigate Command 0x004A required Target Address (as you tested it)
- ZigBeen Pro Stack  ZPS_tsAplZdpMgmtNwkUpdateReq doesn't mention anything about Target address

scan radio channels and report the resultsThe request can be broadcast or unicast to nodes with radio receivers that are configured to remain on during idle periods

So if I'm trying to interpret in the lines, the Target Address will allow to get an Interference Report between Zigate and Target Node (which somehow make sense).

Will investigate in that way.

fairecasoimeme, 2019-06-18:
Zigate doesn't support this cluster and I think this is not the goal.

Zigate Command 0x004A required Target Address (as you tested it)
--> yes, only unicast

ZigBeen Pro Stack ZPS_tsAplZdpMgmtNwkUpdateReq doesn't mention anything about Target address
--> because this is the cluster datas structure. You have to look the good function:

ZPS_eAplZdpMgmtNwkUpdateRequest ( hAPduInst,
                                                 uDstAddr,
                                                 FALSE,
                                                 pu8Seq,
                                                 &sMgmtNwkUpdateReq );

Fred

closed 2019-06-19 by pipiche38