fairecasoimeme/ZiGate #204: Mise à jour firmware impossible

Greg51160, 2019-06-29:
Bonjour,
Le logiciel Flasprogrammer ne parvient pas à trouver ma Zigate TTL.
il voit bien dans la liste le port COM3 qui apparrait à l'introduction dans le port, mais
quand je clique sur Program, il indique "could not read processor id register please check cabling and power"

Merci d'avance pour vos réponses.

Cdt
Greg

fairecasoimeme, 2019-07-01:
Bonjour,
Avez-vous appuyer sur le bouton en insérant la ZiGate sur le port USB ?
Baissez la vitesse à 115200 bauds

Fred

jbvernejoux, 2019-07-03:
Bonjour,

sur Windows 10, pensez aussi à lancer le flashprogrammer ou l'installateur du driver en mode "Run As Administrateur" (pas de double clic sur les EXE mais clic droit sur l'EXE puis "Executer en tant qu'administrateur")
Cela permet d'avoir tous les privilèges système afin d'installer correctement les pilotes et autres opérations modifiant le système.

"Hope it helps"
JB

KiwiHC16, 2019-07-06:
Attention c est tres sensible qu cable USB utilisé. J'ai eu plein de soucis car mes cables n'étaient pas bon. J'en ai trouvé un qui fonctionne nickel, je le gare precieusement.

Autre probleme rencontré, le fichier bin sur un repertoire nfs bien accessible par le browse de fichier mais pas accessible par l'IDE NXP pour le flash.

closed 2020-11-23 by fairecasoimeme