fairecasoimeme/ZiGate #205: Heiman Smoke Sensor.

KiwiHC16, 2019-07-03:
Can't get Cluster 0502, Attribute 0002 reported on USB port. ZiGate is not transparent.

KiwiHC16, 2019-07-03:
#179 is not directly linked but about the same subject.

KiwiHC16, 2019-07-03:
Basically after seting the reporting on the cluster 0500 and attribut 0002 then the Heiman Smoke Sensore report the smoke detection, visible on Air Interface but missing in the message reported by the ZIGate.

e.g. Smoke alarm triggered:
![image](https://user-images.githubusercontent.com/8549674/60605309-92dd5480-9db9-11e9-9fb0-
8766a2958949.png)

Message received from ZiGate:
8102000De734835201050000020019000093

Which give
8102 - message type
000D - message length
e7 - crc
34 - sqn
8352 - sensor address
01 - ep
0500 - cluster
0002 - attribut
00 - success
19 - Attribute Type
0000 - Attribut Size
93 - Quality

As attribut size is zero, and no attribut data provided.

Could it be linked to Attribut Type being 19 ?

pipiche38, 2019-07-03:
Are you with 3.1a firmware or 3.0f?

KiwiHC16, 2019-07-03:
#167

KiwiHC16, 2019-07-03:
3.0f, et visiblement en 3.1a d'apres #167, y a une Modified. Je vais tester.

pipiche38, 2019-07-03:
Yes as on 3.1a the cluster works. Don’t know yet if the doc is update. Îll give you what I have in 30’

pipiche38, 2019-07-03:
Ci-après ce que m'avait fourni akila pour intégrer la Sirène Heiman

Zigate -> Obj 	0x0111 	Write Attribute request IAS_WD (from v3.1a)
                       <address mode: uint8_t>
			<target short address: uint16_t>
			<source endpoint: uint8_t>
			<destination endpoint: uint8_t>
			<direction: uint8_t>
			<manufacturer specific: uint8_t>
			<manufacturer id: uint16_t>
			<Warning Mode: uint8_t>
			<Warning Duration: uint16_t>
			<Strobe duty cycle : uint8_t>
			<Strobe level : uint8_t>

Pour le cluster IAS_WD, c'est la commande 0x0111. pour les paramètres utiles, tu peux t'inspirer du fichier joint pour mettre les bits au bon endroit. Sur ce produit toutes les options ne sont pas utilisées. En gros il prend pas en compte le niveau sonore et la désactivation de la lumière (je crois).
[image]

Sinon n'hésites pas

KiwiHC16, 2019-07-03:
Merci, Ca ressemble beaucoup a ce que je trouve dans la norme. La siren est le probleme inverse du capteur de smoke. L 'un est capteur et l autre Actionneur. Dans le cluster IAS ils appelent ca le WD (Capteur) et le l'ACE (Siren).
Il est possible que le Heiman Smoke Sensor accepte ces commandes pour se comporter comme une sirène mais je n'ai pas d'info la dessus. Je vais faire des tests mais je n'y crois pas trop. On sait jamais avec un peu de chance.

pipiche38, 2019-07-03:
J'ai intégré le Heiman Smoke Sensor, mais j'ai pas cherché à mettre la sirène accessible.
D'abord c'est un device sur batterie, donc je ne suis pas certain qu'on puisse actionner la sirène

fairecasoimeme, 2019-07-04:
En effet, je pense que c'est plus lié au type de données bmap16 plutôt que le cluster

KiwiHC16, 2019-07-06:
Test done with V1.3a and still the same issue. I would tend to say that the issue could be linked to bmap16.
@fairecasoimeme , could you have a look ? Would be nice to have it in next Firmware to support Heiman Smoke detector. For the time being, the smako detecteor is in ZigBee network, we can interrogate it but we can't receive Smoke alarm in case of fire !!!!

pipiche38, 2019-07-06:
Are you sure that this device is able to return a value for 0x502/0x0002 ?

Here is the list of Cluster/Attributes from an Attribute discovery request. There is no 0x0502/0x0002 

As the Bitmap 0x19 it works well on Cluster 0x0500 with Firmware 3.1a

 ```
Attribute Discovery Response - 252e/01 - Cluster: 0000 - Attribute: 0000 - Attribute Type: 20
 Attribute Discovery Response - 252e/01 - Cluster: 0000 - Attribute: 0001 - Attribute Type: 20
 Attribute Discovery Response - 252e/01 - Cluster: 0000 - Attribute: 0002 - Attribute Type: 20
 Attribute Discovery Response - 252e/01 - Cluster: 0000 - Attribute: 0003 - Attribute Type: 20
 Attribute Discovery Response - 252e/01 - Cluster: 0000 - Attribute: 0004 - Attribute Type: 42
 Attribute Discovery Response - 252e/01 - Cluster: 0001 - Attribute: 0020 - Attribute Type: 20
 Attribute Discovery Response - 252e/01 - Cluster: 0001 - Attribute: 0021 - Attribute Type: 20
Attribute Discovery Response - 252e/01 - Cluster: 0001 - Attribute: fffd - Attribute Type: 21
 Attribute Discovery Response - 252e/01 - Cluster: 0003 - Attribute: 0000 - Attribute Type: 21
Attribute Discovery Response - 252e/01 - Cluster: 0003 - Attribute: fffd - Attribute Type: 21
 Attribute Discovery Response - 252e/01 - Cluster: 0004 - Attribute: 0000 - Attribute Type: 18
 Attribute Discovery Response - 252e/01 - Cluster: 0004 - Attribute: fffd - Attribute Type: 21
 Attribute Discovery Response - 252e/01 - Cluster: 0500 - Attribute: 0000 - Attribute Type: 30
 Attribute Discovery Response - 252e/01 - Cluster: 0500 - Attribute: 0001 - Attribute Type: 31
 Attribute Discovery Response - 252e/01 - Cluster: 0500 - Attribute: 0002 - Attribute Type: 19
 Attribute Discovery Response - 252e/01 - Cluster: 0500 - Attribute: 0010 - Attribute Type: f0
 Attribute Discovery Response - 252e/01 - Cluster: 0500 - Attribute: 0011 - Attribute Type: 20
 Attribute Discovery Response - 252e/01 - Cluster: 0502 - Attribute: 0000 - Attribute Type: 21
 Attribute Discovery Response - 252e/01 - Cluster: 0502 - Attribute: fffd - Attribute Type: 21
```

pipiche38, 2019-07-06:
Are you doing the IAS enrolment process for this Device ?

pipiche38, 2019-07-06:
Here after is what I got from the Plugin, when the Heiman Sirene is on Battery mode (un-plugged)

ReadCluster0500 - Security & Safety IAZ Zone - Device: 252e MsgAttrID: 0002 MsgAttType: 19 MsgAttSize: 0002 MsgClusterData: 0010

fairecasoimeme, 2019-07-09:
Salut,

je n'ai que l'alarme Heiman mais elle gère aussi le cluster 0x500. 
Je lance un read attribute sur le cluster 0x500 attribut 0x02.

En retour, j'ai bien une réponse sur l'attribut avec un type de donnée 0x19 et une valeur 0x0000 (j'ai bien 2 en attr size). Sur le sniffer j'obtiens la même chose.

sur ZWGUI : 

10:41:43.189 <- 01 81 02 00 0F 48 05 1F 5C 01 05 00 00 02 00 19 00 02 00 00 9F 03
Type: 0x8102 (Attribute Report)
  SQN: 0x05
  Src Addr: 0x1F5C
  Src Ep: 0x01
  Cluster ID: 0x0500 (Security & Safety: IAS Zone)
  Attribute ID: 0x0002
  Attribute Size: 0x0002
  Attribute Type: 0x19 (Unknown)
  Status: 0x00

 avec un attribute size à 2 et valeur 0

Côté sniffer : 
[image]

Pour moi tout est cohérent ...

Je cherche pour avoir des datas autres que 0 et je reviens vers vous

KiwiHC16, 2019-07-09:
Test avec V3.1a

Sniffer, on a les bonnes info:
[image]

Zigate envoie 
Alarm
8102000D6d0475110105000002**00**1900009c
SQN: 04; Src Addr : 7511; End Point : 01; Cluster ID : 0500; Attr ID : 0002; Attr Status : 00; Attr Data Type : 19; Attr Size : 0000; Data byte list : -><-

Arret alarme
8102000D580575110105000002**00**190000a8
SQN: 05; Src Addr : 7511; End Point : 01; Cluster ID : 0500; Attr ID : 0002; Attr Status : 00; Attr Data Type : 19; Attr Size : 0000; Data byte list : -><-

J'ai toujours attribut size 00 !!!

KiwiHC16, 2019-07-09:
Suis je bien en V3.1a ? 
Si je demande à la zigate elle répond: 030F
Pouvez vous me dire ce que répond la V3.1a ? (Akila cf mon mail).
D'apres le flash tout c'est bien déroulé donc je pense etre en V3.1a.
Si je fais un get Attribut j'ai le même résultat.

fairecasoimeme, 2019-07-09:
A mon avis, ton flash c'est mal passé ou le binaire est pas bon.
la v3.1a renvoie bien 0003031a
Dis moi s'il faut que je te renvoie le binaire.

KiwiHC16, 2019-07-09:
Effectivement quand on est fixé sur une idée on ne voit pas le truc de base.
En allant trop vite j'ai choisi le mauvais bin et tous les tests ont été fait sur la 3.0F...
Donc je reparts de zéro.
Après flash j'ai bien les infos qui remontent:
```
[2019-07-09 15:27:33][AbeilleParser][info]-------------- 2019-07-09 15:27:33: protocolData size(41) message > 12 char: 8102000F8d1dd53a0105000002001900020021cf

[2019-07-09 15:27:33][AbeilleParser][debug];Type; 8102; (Attribut Report)(Processed->MQTT); SQN: 1d; Src Addr : d53a; End Point : 01; Cluster ID : 0500; Attr ID : 0002; Attr Status : 00; Attr Data Type : 19; Attr Size : 0002; Data byte list : ->0021<-
[2019-07-09 15:27:34][AbeilleParser][info]-------------- 2019-07-09 15:27:34: protocolData size(41) message > 12 char: 8102000F8f1ed53a0105000002001900020020cf

[2019-07-09 15:27:34][AbeilleParser][debug];Type; 8102; (Attribut Report)(Processed->MQTT); SQN: 1e; Src Addr : d53a; End Point : 01; Cluster ID : 0500; Attr ID : 0002; Attr Status : 00; Attr Data Type : 19; Attr Size : 0002; Data byte list : ->0020<-
```

KiwiHC16, 2019-07-09:
Je clos donc ce faux problème.

closed 2019-07-09 by KiwiHC16