fairecasoimeme/ZiGate #211: Total Device Limit

schrodingersket, 2019-07-24:
Hello,

I had a quick question for you - I saw that the device limit was updated in ZigbeeNodeControlBridgeCoordinator.zpscfg to 40 child devices and 70 entries in the coordinator neighbor table. I'm looking at increasing that amount, and was curious how you chose the 70 device limit - was that due to a memory limitation on the 5168 chip? I'm working with a network that consists almost entirely of routers, so I'd like to see if it's possible to increase that limit.

Thanks in advance!

P.S. - on a somewhat unrelated note, I have a copy of Stack_zigbee_3.xlsx that I've updated with a few of your changes to various message types. Would you accept a pull request for that?

ISO-B, 2019-07-25:
Device limit is forced due memory on chip. Fairecasoimeme is developing new hardware that has external memory, which should increase device limit. You can read more from: https://zigate.fr/2019/07/03/note-dinformation-juillet-aout-2019/

Stack_Zigbee_3.xlsx hasn't been updated at all on repository. Latest information can be found from: https://zigate.fr/documentation/commandes-zigate/ . If you have something that is missing from there and is still working that way we would like to know so it can be updated to documentation.

schrodingersket, 2019-07-26:
Thank you for the prompt response! I have a quick follow-up question: Is the total limit then 70 devices, or 110 (70 via router, and 40 via coordinator)?

Thanks again.

Xavier-M, 2019-08-10:
Don't want to hijack this question but would or be possible to clarify this 40/70 limit? Does it mean 40 battery devices + 30 routers? Or 110 total?

And what happens when you try to add a device after that? A specific message in the logs?

Thanks a lot :-)

schrodingersket, 2019-08-10:
I'm actually working on getting a test environment set up to try this. I'll update as I find out more. (:

fairecasoimeme, 2019-08-14:
40 devices with direct link with ZiGate and 30 more if you have some router devices. Router device can stock some devices and route to ZiGate.
So, 70 devices max with ZiGate.

Unfortunatly, ZiGate don't say nothing when you attemp to join a new device when it's full. No message when ZiGate is full too.

Perhaps, we have to enhance ...
(We work too to improve with add a memory flash)

schrodingersket, 2019-08-14:
I'd be happy to implement this notification functionality, if you'd like.

schrodingersket, 2019-08-14:
Following up on an earlier thought: do devices connected directly to the coordinator take up more memory than those connected via a router? If so, would decreasing the number of devices allowed to connect to a coordinator increase the total number of connected devices if the network was predominantly comprised of routers?

ISO-B, 2019-08-15:
Good topic with information about table sizes: https://community.nxp.com/thread/500642

It is possible to check table sizes any time. Probably the best time would be when permit join command is received. Then command could respond with status to permit join and also how many devices can join.

pipiche38, 2019-08-19:
This sound good if we could get a Zigate command to query those Table.

pipiche38, 2019-08-20:
> 40 devices with direct link with ZiGate and 30 more if you have some router devices. Router device can stock some devices and route to ZiGate.
> So, 70 devices max with ZiGate.
>
> Unfortunatly, ZiGate [...]

What seems to be strange is that Zigate is not able to add the device to the routing table, however the Device Annouced message is send back to the SerialLink (plugin).
Cannot we stop this message ?

pipiche38, 2019-11-28:
see #264