fairecasoimeme/ZiGate #212: AttributeDiscoveryResponse and complete status

bghes, 2019-07-24:
AttributeDiscoveryResponse always send complete 1 which not permit to get the end of the attribute list.
Is-it a bug ?

ISO-B, 2019-07-25:
I can take a look. Could you provide logs?

pipiche38, 2019-07-25:
@bghes It might be good that you provide also information on the Firmware Version.
I'm using this feature and never see any issue ! Can it be linked to the device itself ?

bghes, 2019-07-27:
Hello,
I use the last firmware 3.1a with a driver i have developped. 
In the log i print the complete status of each attribute discovery response.
[trace-2.txt](https://github.com/fairecasoimeme/ZiGate/files/3438265/trace-2.txt)
[trace.txt](https://github.com/fairecasoimeme/ZiGate/files/3438249/trace.txt)

Thanks

pipiche38, 2019-10-09:
On my case I'm not using this flag. I'm collecting all Attribute discovery responses to construct the full list of Cluster Attribute for the particular device.

I'm doing that , most-likely because indeed the flag is always set to 1 ;-)

pipiche38, 2019-10-10:
I do confirm that the Flag seems to be alway set to 1.
From what I saw in the code, it is just passing the flag from the Zigbee stack.

So I would recommend you not to take the flag in consideration, but more to compile every attribute you receive.

pipiche38, 2020-11-22:
Closed as a feature

closed 2020-11-23 by fairecasoimeme