fairecasoimeme/ZiGate #213: APS data confirm fail but not asking anything

biau, 2019-07-25:
Hi,
Using zigate on hass with doudz lib I can find some error message for my motion sensors:
```
2019-07-25 12:53:03 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10 X\xb4\xcf\xcc\x02\x11\x02\x10\x02\x10\xff\x02\x12\x02\x10B\x02\x10\x13\x10\x02\x11!\xa9\x02\x1b!\xa8\x13$\x02\x11\x02\
x10\x02\x10\x02\x10\x02\x10!\x14\x02\x10 T\x8d\x03'
2019-07-25 12:53:04 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-07-25 12:53:04 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'b4cfcc010000ff0200420013100121a90b21a8132401000000002114002054'
2019-07-25 12:53:04 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:180, addr:cfcc, endpoint:1, cluster:0, attribute:65282, status:0, data_type:66, size:19, data:100121a90b
21a8132401000000002114002054, lqi:141
2019-07-25 12:53:04 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
2019-07-25 12:53:04 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-07-25 12:53:11 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x87\x02\x12\x02\x10\x02\x1e\x98\xf0\x02\x11\x02\x11\x02\x12\xcf\xcc\x02\x10\x02\x10\x02\x10\x02\x10\x02\x10\x02\x10\xe2\x02\x10\x03'
2019-07-25 12:53:11 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-07-25 12:53:11 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8702: b'f0010102cfcc000000000000e2'
2019-07-25 12:53:11 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:226, lqi:0
2019-07-25 12:53:11 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:226, lqi:0
2019-07-25 12:53:11 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
```
this happens every 55 minutes as you can see here:
```
019-07-25 06:21:37 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:223, lqi:0
2019-07-25 06:34:33 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:242, lqi:0
2019-07-25 07:16:38 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:47, lqi:0
2019-07-25 07:26:07 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:59, lqi:0
2019-07-25 08:11:42 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:121, lqi:0
2019-07-25 08:18:34 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:132, lqi:0
2019-07-25 09:06:43 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:192, lqi:0
2019-07-25 09:13:04 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:200, lqi:0
2019-07-25 09:59:04 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:13, lqi:0
2019-07-25 10:08:08 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:24, lqi:0
2019-07-25 10:54:08 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:80, lqi:0
2019-07-25 11:03:11 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:90, lqi:0
2019-07-25 11:49:09 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:148, lqi:0
2019-07-25 11:58:11 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:158, lqi:0
2019-07-25 12:44:10 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1abc, sequence:208, lqi:0
2019-07-25 12:53:11 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:cfcc, sequence:226, lqi:0
```

Doudz tells me that error code is: MAC_ENUM_TRANSACTION_EXPIRED 0xF0 Pending transaction has expired and data discarded but we can't find any request before receiving this response.

Both devices are xiaomi motion sensor V1.
And they all respond with no issue to motion events and I can turn light on with automation.

zigate firmware is 3.0f.
original issue: https://github.com/doudz/homeassistant-zigate/issues/60

pipiche38, 2019-07-26:
same as #181

pipiche38, 2019-07-29:
If those 2 devices : 1abc and cfcc are indeed Xiaomi sensors, this means that the plugin is sending a request  to those, which doesn't make sense as Sensors are battery based devices and are not listening on Idle.

doudz, 2019-07-29:
But looking in the log, there's nothing sent to the sensor

doudz, 2019-07-29:
The only thing could be during device discovery ,the plugin could try to configure a periodic reporting on some general attributs.
Generally it fails but maybe it generate a periodic failure instead ?

closed 2020-11-23 by fairecasoimeme