fairecasoimeme/ZiGate #214: What is status 0x15 in 0x8000 message

pipiche38, 2019-07-26:
I'm getting a Status code of 0x15 with 0x8000 after sending a command.

I didn't try to reproduce with ZGUI, but the way I'm doing is : 

1- Pair a device on the Zigate
2- Send a On/off to the device in order to check everything is in order
3- un power this device
4- Send On to the un-powered device
    -> You'll recevive 0x8000 with status 0x00 and 0x8702
5 - send On to the un-powered device (again)
    -> You'll recevive 0x8000 with status 0x00 and 0x8702
6- send On to the un-powered device (again)
    -> You'll recevive 0x8000 with status 0x00 and 0x8702
7- send On to the un-powered device (again)
    -> You'll recevive 0x8000 with status 0x15

pipiche38, 2019-07-30:
It looks like it could be that one, 

src/sdk/JN-SW-4170/Components/ZCIF/Include/zcl.h

E_ZCL_ERR_ZTRANSMIT_FAIL

pipiche38, 2019-08-02:
After invetsigation I have seen that 0x8000 can return 2 ZCL error code
E_ZCL_ERR_ZBUFFER_FAIL 0x14
E_ZCL_ERR_ZTRANSMIT_FAIL 0x15

pipiche38, 2019-08-02:
And when looking in the Source code when is E_ZCL_ERR_ZTRANSMIT_FAIL , it happens when there is an error with ZPS_eAplAfUnicastDataReq . Why not returning the return codes of ZPS_eAplAfUnicastDataReq  instead of forcing to 0x15 ?

CC: @ISO-B

fairecasoimeme, 2019-08-07:
I find this :
https://community.nxp.com/thread/450656
Could you confirm there is the same issue ?

pipiche38, 2019-08-08:
Not necessarily related to an APS Failure.

For me it is more simple, it would be better to return the real zigbee stack pro code rather than -15

pipiche38, 2019-08-08:
> 
> 
> I find this :
> https://community.nxp.com/thread/450656
> Could you confirm there is the same issue ?

@fairecasoimeme , I have read once more the issue you pointed out and honestly I do not understand the response that the person got.
For me based, on what I have experienced and what you have written in an other issue, indeed if you retry the same data frame , you get E_ZCL_ERR_ZTRANSMIT_FAIL.

However I still see from the Zigate Source code, that this E_ZCL_ERR_ZTRANSMIT_FAIL is set by Zigate firmware and not the underline API. So it would be better to have the API status instead

pipiche38, 2019-08-21:
UP thanks!

pipiche38, 2019-08-23:
I beleive this one is impacted by #218 as now it returns the result of ZPS_eAplAfUnicastAckDataReq() and not a fix E_ZCL_ERR_ZTRANSMIT_FAIL

closed 2019-08-23 by pipiche38