fairecasoimeme/ZiGate #217: Only 1 line in TestGUI app

pasoad, 2019-07-31:
Hi everyone,

My zigate is not working for now and i tryed to re-install it [by following this](https://zigate.fr/documentation/mise-a-jour-de-la-zigate/) and after, testing the communication with the TestGUI app [here](https://zigate.fr/documentation/tester-la-zigate-usb-v1-1/) but i've got some trouble when i clik on "Get Version".

The answer is : `00:52:39.750 -> 01 02 10 10 02 10 02 10 10 03 `

The guide say's it has to return 3 lines... 

Both light (red and blue) are power on, is that normal ?

Can you help me with this please ?

pipiche38, 2019-08-01:
It looks like you don't get any response from the Zigate.

If I understood you have done the Firmware update and this has been positive.

When you place the Zigate on the USB port and you run ZGUI, after setting the speed and opening the port, what is the Status of Zigate. Do you have any leds ?

pasoad, 2019-08-03:
thanks for your reply :)

i have both led power on.

but now, the test still give me the same result but the zigate works in jeedom.

i don't understand why but it's ok...

fairecasoimeme, 2019-08-07:
I think this is a windows driver problem.
Please try to reinstall driver even if you have the red usb module.

Fred

letroll, 2020-08-01:
it's also only 1 line, with bad baudrate. it's should be 115200

closed 2020-10-07 by fairecasoimeme