fairecasoimeme/ZiGate #219: Limitation of 5 groups seems vefry low

pipiche38, 2019-08-02:
Since Firmware 3.1f , limitation of not more than 5 Groups makes quiet challenging to configure groups in a real environment.

Is there any possibility to increase such Number what else could we put in balance ?
Can it be something configured at Erase PDM time where you would specify Number of Groups vs. number of devices or whatever parameters which came in the balance ?

ISO-B, 2019-08-02:
Groups and devices are defined at build so there is no way to switch those on fly. Akila has been working for new hardware which will feature external eeprom. With new hardware we should be able to increase number of devices and groups.

Meanwhile and with current hardware only solution is to find out how many devices we would lose if we for example would increase number of groups to 10 or 15.  If I recall correctly it is not that many devices that we would lose, probably something like 1 device per 2-3 groups or so.

pipiche38, 2019-08-05:
@ISO-B how shall we consider the limitation of 5. Is that the total number of groups in the Zigbee network or is that simply limited to the number of Groups where ZIgate has also a membership ?

For instance if I have an Ikea Tradfri Remote paired with 5 bulbs, is that also considered as a group ?

ISO-B, 2019-08-05:
It is number of groups that ZiGate is member. There can be more than 5 groups in network, but ZiGate can't be member in more than 5.

fairecasoimeme, 2019-08-07:
Yep, @ISO-B says all

closed 2019-08-08 by pipiche38