fairecasoimeme/ZiGate #220: Konke Motion integration

pipiche38, 2019-08-03:
There is a strange behaviour of the Konke Motion which is on a regular basis is sending a 0x8102 Individual Attribute response and in such is creating a False Positive as the Alarm1 is set to 1.

pipiche38, 2019-08-03:
Every 15' 
2 times 0x8102
2 times 0x8401

pipiche38, 2019-08-05:
I would like also to confirm that Enrollement is also made, and I got that confirmation also every 15' !

fairecasoimeme, 2019-08-07:
I test my motion sensor on my ZiGate and I come back to you to tell my results

fairecasoimeme, 2019-08-09:
I test my device but I didn't reproduce the issue.

My sensor send only when it detects moves. only 1x 0x8401

Could you test directly with ZWGUI ?

Fred.

pipiche38, 2019-08-09:
Will do and revert

pipiche38, 2019-08-09:
Ci-joint le problème avec ZGUI

Trames reçues à 15:56, 16:04, 16:20 alors que le Motion sensor est dans sa boite dans une armoire
 
[image]

fairecasoimeme, 2019-08-09:
C'est vraiment bizarre, je n'ai pas du tout ça de mon côté.
Tu as quelle version de firmware ? la v3.1a ?

pipiche38, 2019-08-09:
Oui, je vais regarder de plus pret la phase d'appairage et notamment il y a des bindings qui sont fait et un configureReporting, qui peuvent peut-etre agir :-(

Je cherche de mon coté et je te dis quoi ;-)

fairecasoimeme, 2019-08-09:
A oui s'il y a des bindings, c'est surement ça.

pipiche38, 2019-08-09:
Je ferme l'issue, désolé j'aurai du poussé les tests un peu plus loin.
Le problème semble etre lié au Binding/Configure Reporting

closed 2019-08-09 by pipiche38