fairecasoimeme/ZiGate #222: Where's SrcAddr in 0x8040 ?

doudz, 2019-08-05:
In the last firmware 3.1a changelog I see : 
- Add SrcAddr to 0x8040 command (MANAGEMENT_LQI_REQUEST) 

But looking at doc and source code I don't see any change, am I missing something ?

doudz, 2019-08-05:
Ok, just realize it's 0x804E and not 0x8040, the changelog should be updated ;)

doudz, 2019-08-05:
Doc for command 0x0807 is missing and wrong mention to 0x8806

pipiche38, 2019-08-05:
MsgSrc has been also added to 0x8040. It is added at the end of the Payload, so in comparison of the version 3.0f it is uint16 after the Neighbour Table list

doudz, 2019-08-05:
@pipiche38 are you sure it's 0x8040 ? I think is 0x804E

pipiche38, 2019-08-05:
@doudz indeed:

0x804A and 0x804E have been updated in order to provide Source Address

pipiche38, 2019-08-05:
Just check the documentation, and it looks correct: 
for 0x804E and 0x804A https://zigate.fr/documentation/commandes-zigate/

I strongly recommend to remove this page: https://zigate.fr/wiki/commandes-zigate/ as this one is obsolete. Despite the fact that I found it more readable than the new one !

@fairecasoimeme

doudz, 2019-08-05:
Sure, but I'm talking about https://zigate.fr/documentation/commandes-zigate/ which is not up to date ;)

fairecasoimeme, 2021-03-10:
doc updated
add 0x0807 command 
0x8806 and 0x8807 command don't exist in the protocol

closed 2021-03-10 by fairecasoimeme

tcharp38, 2021-03-10:
@fairecasoimeme 
I did not check them yet but thanks for all these doc updates.