fairecasoimeme/ZiGate #225: 0x8024 with status 6 ?

doudz, 2019-08-11:
A user report a problem ,0x8024 return status 6, what does it mean ?

pipiche38, 2019-08-11:
From what I saw in the Source,

0x0024 --> APP_vControlNodeStartNetwork --> BDB_eNfStartNwkFormation()

BDB_E_ERROR_COMMISSIONING_IN_PROGRESS

@fairecasoimeme to confirm

doudz, 2019-08-11:
Hum, and what should we do? Just wait a little and retry command 0x0024?

ISO-B, 2019-08-15:
Is user trying to start network as coordinator not end device or router? Is this after fresh install?

doudz, 2019-08-15:
The user got that on first start after upgrading to 3.1a

pipiche38, 2019-08-21:
@doudz, this message is indeed coming from 3.1a, where 0x8024 messages have been normalized (see #74) .

0x06: Commissioning in progress. If network forming is already in progress

For the meaning, may be @fairecasoimeme can comment

closed 2020-11-23 by fairecasoimeme