fairecasoimeme/ZiGate #227: Device removal 0026 doesn't behave as expected #29

pipiche38, 2019-08-19:
I just did some test today, and if you set the Zigate IEEE as the 1st IEEE, the device is not removed from the Zigate. Instead if you put TWICE the IEEE of the End Device then the device is removed from the Zigate.

I did the test with an Xiaomi Plug

Why do I say the removal is complete or not is because

if I use twice the device IEEE with command 0026, after submission of that one, whatever I'm doing with the Xiaomi plug, I do not see any more message on Zigate from this Device

If I use Zigate IEEE, and Device IEEE, after submission of the command I still receive messages from the Xiomi Plug

pipiche38, 2019-08-19:
If I read the documentation, what I understand is the first parameter is the IEEE of the router we want to do the removal. 

> This function can be used (normally by the Co-ordinator/Trust Centre) to request another node (such as a Router) to remove one of its children from the network (for example, if the child node does not satisfy security requirements).
> The Router receiving this request will ignore the request unless it has originated from the Trust Centre or is a request to remove itself. If the request was sent without APS layer encryption, the device will ignore the request. If APS layer security is not in use, the alternative function ZPS_eAplZdoLeaveNetwork() should be used.

(1) I don't think this is up to the Application layer to know the exact Topology of the network and to which router the end device belongs too
(2) By putting twice the IEEE it might be just a side effect
(3) If removing a end device, then the 1st parameter should not be the IEEE but it could be a Router the Edn device is connected to.

pipiche38, 2019-08-27:
@fairecasoimeme , @ISO-B , I think this will become a burning issue, as I see more and more user reaching the limit of Zigate in terms of number of direct device connected to it.

I also would like to bring one topic here which is removing battery based devices. In the following comment: https://github.com/fairecasoimeme/ZiGate/issues/29#issuecomment-438690133 @fairecasoimeme you are mentioning that Zigate must receive the leave command from the device itself. While I can understand this , I'm also thinking that we should be able to force the process and simply tell Zigate to forget and unpair. This is BTW what is happening when the Zigate doesn't have contact from a device since a long time. Counting on the Leave command from the end device looks to me problematic, especially if this device is not in the network anymore. Taking into account that I undertsood you have increase the contact Timeout (due to the Xiaomi issue) to a long time.

fairecasoimeme, 2019-08-27:
I already worked on this subject and the only way, for me, is to use 0x26 command or 0x4C command.
I don't know if it's due to security process but only device can leave not ZiGate.
May be you could play with ZiGate memory and delete entry of the table but I don't know if it's really good. Perhaps, we have to decrease timeout value (finally there is no impact on Xiaomi device)

pipiche38, 2019-08-27:
0x4C, is that a new command ? What is the difference with 0x47 ?

From what I see in the NXP doc, indeed 0x4C seems to be much better, as we can manage - or at least - request the child to rejoin

I would be interested in your reading and understanding of that:

> 
> This function can be used to request a node to leave the network. The leaving node
> can be a child of the requesting node or can be the requesting node itself (excluding
> the Co-ordinator).
> The node being asked to leave the network is specified by means of its IEEE (MAC)
> address (or zero, if a node is requesting itself to leave the network). You must also:
> Use the parameter bRemoveChildren to specify whether children of the leaving node
> must leave their parent - if this is the case, the leaving node will automatically call
> ZPS_eAplZdoLeaveNetwork() for each of its children. This parameter must always be
> set to FALSE when the function is called on an End Device (as there are no children).
> Use the parameter bRejoin to specify whether the leaving node must

It is not really clear for me, If I want to remove for instance a Node (which is a router) and I want the Childs to rejoin immediatly. I would say that bRemoveChildren to FALSE and bRejoin to TRUE should make the trick, but I'm wondering if the Node I want to remove is not going to rejoin as well !

fairecasoimeme, 2019-08-27:
Hi,
Oups, indeed the 0x4C is not documented...
With this command you add following datas : IEEEaddress then if rejoin or not and after (if it's a router) if you want to remove children or not.

I think in your case, you have to simply change bRemoveChildren to FALSE and bRejoin to FALSE. The Ieee address leave network but not its children. That's all. childs stay to network and refind a new route if necessary.

Fred

pipiche38, 2019-08-27:
Fred, So Rejoin is applicable to the node I want to remove, and if RemoveChildren is False then they will reconnect by themselve.

This is indeed the right way to remove a node

fairecasoimeme, 2019-08-27:
Yes I think it's a good way.
Fred

closed 2019-08-27 by pipiche38