fairecasoimeme/ZiGate #228: Zigbee Version Support

ruimarinho, 2019-08-20:
Aqara is apparently [launching all-new T1 sensors](https://homekitnews.com/2019/07/08/aqara-unveils-new-m2-homekit-hub-switches-locks-cameras-and-more/) with support for Zigbee 3.0.

I always thought ZiGate was Zigbee 3.0-compliant so, in that case, what version of Zigbee are current Aqara sensors running? Is there any expected advantage of these new sensors (e.g. current sensors forcing older Zigbee protocol on ZiGate)?

pipiche38, 2019-08-20:
For me the Xiaomi are current ZHA 1.2 compliant having them been Zigbee 3.0 might help and smooth integration and reliability.

ruimarinho, 2019-08-24:
Thanks @pipiche38. I think I will way for the updated hw revision then. Curious on what the improvement will look like!

closed 2019-08-24 by ruimarinho