fairecasoimeme/ZiGate #229: Data Type 0x25 uint48 seems to be wrong

pipiche38, 2019-08-20:
When receiving an attribute of Data Type 0x25, the corresponding length is 0x0008, while it should be 0x0006 , as 0x25 is Unsigned 48 Bits 6 x 8 bits.

In addition the Data attributes (which has a length of 8) seems to be truncated and only the 4 first bytes are correct !

The issue is reported while receiving Cluster  0x0101 Attribute 0x0508 of the Xiaomi Vibration.

The end result is that we are unable to compute the Vibration orientation !

schrodingersket, 2019-08-20:
Related to #223 perhaps? (:

pipiche38, 2019-08-21:
You might be right, as I did further investigation and indeed even for Data Type 0x2a we got a size of 0x0004 instead of 0x0003 as we are expecting a U24bits !

It was however fixed in 3.0f !

@fairecasoimeme

fairecasoimeme, 2019-08-23:
It's normal, I cast Uint48 with Uint32.
But you're right, there are inconsistency between size and datas. I'll fix that.
thanks

pipiche38, 2019-08-24:
Is there any impact on this fix to the 0x2A Data Type ?

fairecasoimeme, 2019-08-24:
No :( no change for 0x2A. How could I test ? do you know a device which use this data type ?

pipiche38, 2019-08-24:
Basically Cluster 0x0702 / Attribut 0x0400
As of today on Firmware 3.1a

Here is what I'm getting:

Attribute Type: 0x2a
Attribute Size : 0x0004
Attribute Value: 0000011d

And that is where I'm concern, as If I take the logic you put now, I would expect

Attribute Size: 0x0003
Attribute Value: 00011d

Which is quiet different

pipiche38, 2019-08-27:
I'll try to test it and revert .
Main issue is the Salus Plug I have is on my prod network ... So I have some logistic to be done

closed 2019-09-01 by pipiche38

ISO-B, 2019-09-02:
@pipiche38 Did you managed to work solution to this?

pipiche38, 2019-09-02:
The 0x25 uint48 is indeed fixed by 3.1b I got to test. So it is good.

aargollo, 2020-09-10:
Hi Guys!
I would like to check if somebody have tested the last Zigate Version (v3.1.c) Data type 25 with Attribute CurrentSummationDelivered (0x0000)?
I observed that the zigate show 8 bytes length , over Data Type 25, and is discarding the measurement important information. Alway represents all those bytes as 0x00. As the PIC attached bellow.

[image]

I got messages over the air, via wireshark and I observed it:
**ZCL Command :** 0x0a - Report Command Attribute 
Attribute filed:
**Attribute:** Current Summation Delivered (0x0000)
**Data Type:** 48-bits Unsigned Integer
**Data:** Measurement value (6 bytes).
I believed that Data Type 25 should be have this format: .... 25, **0x2, 0x10, 0x2, 0x16,** + **6 bytes** + RSSI Byte and EOT (0x03).

According Zigate Protocol (v.3) Report Attribute Response should be have:
<Attribute data   type: uint8_t>
<Size Of the attributes in bytes:   uint16_t>
<Data byte list : stream of   uint8_t>

[image]

pipiche38, 2020-09-10:
Do you have this problem only with ZGUI ? 
As I do not see any issue with the current 31c which report perfectly on Data Type 0x25

aargollo, 2020-09-10:
Hi my friend. It´s a good question, I didn´t do it before, but today I did double check in the ZGUI and I observed the same behavior as I posted yesterday. As you can see bellow:

 according Wireshark the data value from Data Typ2 25 - Attribute CurrentSummationDelivered (0x0000), the data value (6bytes), from message (SQN e9 - 233) is: **2c02**
[image]

But in the ZGUI the same message (SQN e9 - 233) all those bytes in the Data type 25 are all those bytes as zero (0x00).
[image]

pipiche38, 2020-09-10:
It might be the case that ZGUI is not good to handle the Data Type 0x25 which was fixed in 31b/31c firmware. Have you tried with a python library or something else ?

aargollo, 2020-09-10:
HI Fred,
I have include the 3.1.c version.
I got the packages from my Node-red interface and I saw the same results,
all those 8 bytes with 0x00 value.

Best Regards,
Alex

aargollo, 2020-09-11:
HI guys, Sorry. I don´t know why, but after new tests I observed that my Zigate had inside a old firmware, since I did the 3.1.c update.  and I could check that everything is ok. I have tests Current Summation Delivered (Data type 25- 48 bits) and Instantaneous demand (data type 42 - 24 bits).
**Data flow:**
[image]

**OTA log via wireshark**

 
--

commit e51ff00f (2019-08-23): fix #229