fairecasoimeme/ZiGate #233: Neighbored setup

ruimarinho, 2019-08-24:
Hi,

I have a fairly small zigbee network setup which has had one little problematic sensor (add 9610 in the picture). With the PL2303 module, the sensor would disappear after a couple of days. I just got CP2102 and now the sensors appears to be always connected but I stop receiving messages a couple of days after starting the network.

This is the network setup:

[image]

The LQI doesn't seem bad enough (LQI 72 or 26%) to cause this. Anyway, I tried adding a signal repeater from IKEA physically next to this sensor which you can see has very good reachability to zigate. Is there any reason why `9610` wouldn't use this repeater instead to improve its LQI?

Maybe its current LQI would become worst by performing another hop? Any feedback appreciated.

fairecasoimeme, 2019-08-27:
Hi,
It's difficult to manage network routing because we can't control zigbee algorithm. When there are routers and devices with medium quality, it's possible that device don't "hang" to router. The best way to force (but with time, device can go out from router) is to pair router to zigate first, and after, pair device very close to router.

ruimarinho, 2019-09-14:
Do I need to unpair device and pair again near router or I can simply force re-pair?

fairecasoimeme, 2019-09-16:
Try to unpair and pair again. what is for you a force re-pair ?

ruimarinho, 2019-09-16:
Put zigate in permit mode and pair device using reset button on the device for example.

fairecasoimeme, 2019-09-16:
Ok, so It's the same things. With device, reset and pair is the same button. the first do a leave the second do a re-pair.
You can do with this way.

closed 2020-10-07 by fairecasoimeme