fairecasoimeme/ZiGate #235: Zigate limitation

pipiche38, 2019-09-03:
While the Zigate limitation is 70 devices with a max of 40 directly connected to itself and 30 through routers, is there a way for Zigate to push devices to routers ?

For instance, what I see is that I have a perfectly meshed network, however , the devices remains directly connected to Zigate. Which make sense as it does reduce the risk of loosing network if the LQI is good enough. 

However due to the 40 devices limitation, end users are stuck .

Can the limitation of 40 + 30 be reconsider in 70 directly connected as we do really really cares above. As today, I do consider that today's Zigate limitation is in fact 40 Devices - and not 70 -

deennoo, 2019-09-03:
I'm facing this issue too, i have to add 2 more devices and i can't do it directly.

I have to move some xiaomi device from my zigate to my xiaomi gateway and then i can add some other device from other brand.

Some illustrations : 

[image]

[image]

pipiche38, 2019-09-04:
@deennoo problem of moving Xiaomi devices is that if there are simple sensor (so not Routers) we cannot easily remove them from the Zigate!
Again from my end I see the constraint of 40 direct devices to Zigate as the main constraint without the possibility to push some of them to Routers !

ruimarinho, 2019-09-04:
@pipiche38, can you explain that better? Does that mean Xiaomi battery powered devices can only connect to Zigate directly, not through separate router?

pipiche38, 2019-09-04:
@ruimarinho , it can connect to a router, but if there is a little possibility to go to the zigate, what I have observed is the device connect to Zigate. And unfortunately there is no way to control that.

Then my main concern is more on the Routers, which will keep a route to the Zigate even if it is not the best, and in that case it will consume one slot out of the 40 !

KiwiHC16, 2019-10-29:
This is up to the device to decide where to connect based on radio quality. We can try to push a Leave with Rejoin to the Device with a bit of chance it will reconnect to a router. First not easy to send the request to battery device without physical action and probabily to get it to a router is not 1.
I guess the best will be Zigate with external prom to increase number of device.

GeitjeG, 2019-11-12:
For me this would be a request also. I do not have 40 devices yet, but will have in the future. Using the GU10 spots my number increases quick...

pipiche38, 2019-11-27:
> This is up to the device to decide where to connect based on radio quality. We can try to push a Leave with Rejoin to the Device with a bit of chance it will reconnect to a router. First not easy to [...]

I do not agree the problem is the +30 devices which can be connected via routers, this is creating a potential network instability in case of router power outage. 
Increasing the prom size will simply delay the time when to have the problem, so I strongly recommend to keep the same limit, which is today 40 devices , this will make the setup reliable (while the +30 is a pure internal mecanism, on which nobody has control)

closed 2020-06-29 by pipiche38