fairecasoimeme/ZiGate #236: Print Log debug on UART1

minhtan58, 2019-09-09:
Hi everyone
Now. I'm working with ZigGate source code.  But I cant'n print log to UART1

#define UART_DEBUGGING                                            TRUE
#define TRACE_APPSTART                                              TRUE
#define TRACE_MIGRATION                                           TRUE

Can you help me? Thanks !!!!

ISO-B, 2019-09-09:
https://github.com/fairecasoimeme/ZiGate/blob/a8730b144a282a2a47308ddafa947632e63b230e/Module%20Radio/Firmware/src/ZiGate/Build/ZigbeeNodeControlBridge/Makefile#L64-L66

Change TRACE to 1 and DEBUG to UART1 and then rebuild.

pipiche38, 2020-11-22:
Please check with zigate project
Can be closed

closed 2020-11-23 by fairecasoimeme