fairecasoimeme/ZiGate #237: Build Zigate

minhtan58, 2019-09-14:
Hi everyone
I get a problem When I build ZiGate v3.1a (final) project.
Use Zigate from https://pypi.org/project/zigate/ to connect to Coordinator,

Use ZiGate_coordinator_JN5169.bin (in forder bin) it connect ok.
Build JN5169-115200
https://user-images.githubusercontent.com/34535797/64834116-d3075180-d60a-11e9-8e56-b740595e46d9.png
=> I can't open port to connect Coordinator
Can you help me? I don't know what reason !!!

minhtan58, 2019-09-16:
[image]

[image]

closed 2019-09-16 by minhtan58

doudz, 2019-09-16:
What was the problem ?