fairecasoimeme/ZiGate #238: List all paired Devices managed by Zigate

pipiche38, 2019-09-18:
I beleive that there are some tables which contains all of the paired devices known by Zigate as an active device.
It would be great to have an API allowing us to list those devices ( IEEE, NwkID and eventually other usefull information for above layers).

-> This will help to check consistency between what Zigate has and what aboive layers are aware
-> This will give an exhaustive list of Devices against the Max Zigate Limit

As I understood command 0x0015 is not giving such exhaustive list , but much more the most active list, which is not enougt in front of the Maximum Limit.

pipiche38, 2019-09-18:
@ISO-B was refering to such table in an other post in regards to Zigate limits

KiwiHC16, 2019-10-29:
From what I recall, such table doesn't existe. But based on association behavior I'm sure that TC has a table to manage authentication. I haven't been able to find any information about it...

closed 2020-05-08 by pipiche38