fairecasoimeme/ZiGate #241: Zigate Box 3D Model

ruimarinho, 2019-10-02:
Hi @fairecasoimeme,

I have bought the new TTL module and as you mentioned, the plastic enclosure is slightly bigger than it should.

I have a friend who can do the 3D printing for me of an adjusted enclosure. Would you mind sharing the model with the community?

Thank you,

fairecasoimeme, 2019-10-09:
Hi,
you can find here
https://www.hammfg.com/fr/part/search?filter%5Bq%5D=1551USB3

fred

pipiche38, 2019-11-28:
can be close

ruimarinho, 2019-11-30:
Indeed! Thank you @fairecasoimeme. First prototype last week is looking nice :) it offers a much better fit with only 3mm less in length!

[image]
[image]
[image]

closed 2019-11-30 by ruimarinho

doudz, 2019-11-30:
I'm going to print a new box too to with a button to be able to use the flash mode without disassembling the box