fairecasoimeme/ZiGate #242: Is cluster 0xFC01 (Legrand) is implemented ?

pipiche38, 2019-10-06:
From what I understood:

0xFC01 / 0x0000 is the DIM value
0xFC01 / 0x00001 is the Led state

Smanar, 2019-10-06:
Exactly

pipiche38, 2019-10-09:
Probably not needed, as the Write Attribute is enought!

closed 2019-10-09 by pipiche38

pipiche38, 2019-10-12:
@Smanar seems that Legrand is playing a bit

Here are the decoded/sniffed 0xFC01 attributes and what they do:

```
'Dimmer switch w/o neutral':  { 'EnableLedInDark': '0001'  , 'EnableDimmer': '0000'   , 'EnableLedIfOn': '0002' },
'Connected outlet': { 'EnableLedIfOn': '0002' },
'Shutter switch with neutral': { 'EnableLedIfOn': '0001' },
'Micromodule switch': { 'None': 'None' }}

Seems that Attribute 0x0001 and 0x0002 depends on the Device itself !
```

pipiche38, 2019-10-12:
Please do note also that when the Zigate is in Pairing mode, then all Legrand devices (main powered) will get their Led Green, otherwise they will follow the 0xfc01/EnableLedIfOn and EnableLedInDark)

Smanar, 2019-10-12:
Yep, all LED go to green ^^.
But you can too change the color too using the identify zigbee command, and seelct the color you want.

Have you see the last legrand device, une "sortie de cable" than can do on/off but "fil pilote" too. Just using neutral + fire wires ?