fairecasoimeme/ZiGate #243: Appairage device Legrand (après avoir été mis en contact avec le Hub/Prise connectée)

pipiche38, 2019-10-06:
J'observe un changement de comportement -pas glop du tout- quand un device Legrand/Netatmo est appairé à la Box Legrand/Netatmo.

En effet alors que l'appairage et le fonctionnement est parfait (ou semble etre parfait) sur un device vierge, lorsque celui-ci a été appairé (vraisemblablement upgradé) à un Hub Netatmo/Legrand, l'appairage sur la Zigate est quasiment impossible. En fait au bout d'une dixaine de secondes, le device envoi un Leave et disparaît !!

cf: #59 , #242

Smanar, 2019-10-06:
Je pensais d'apres ton PM que ca avait finallement marché ?

pipiche38, 2019-10-06:
> Je pensais d'apres ton PM que ca avait finallement marché ?

Correct, ca fonctionne, mais cela ne me semble pas une façon très propre de faire un appairage. Donc pour moi il manque quelque chose quelque part ;-)

Smanar, 2019-10-06:
ha, avec la coupure secteur ^^ ? Ca ne marche pas sans ca ?

pipiche38, 2019-10-06:
> ha, avec la coupure secteur ^^ ? Ca ne marche pas sans ca ?

Pas la coupure, mais un autre truc du meme genre

pipiche38, 2019-10-10:
A priori, il semble que le problème se manifeste également sur la box Legrand ! A creuser auprès du support Legrand.

pipiche38, 2019-10-17:
Après investigations ( test + lecture documentation) la procédure d'appairage Legrand est bien

- Bouton Reset ( > 8 secondes ) afin d'avoir un full reset
- relâcher et re-cliquer pour confirmer.

Donc pas besoin de  spamer avec plusieurs clicks après avoir relâcher. Un seul clique est suffisant (il est considéré comme clique de confirmation)