fairecasoimeme/ZiGate #246: Impossible to re-include deleted Xiaomi detectors

biscwii, 2019-10-10:
Hi there
 
I have an issue that I hope someone here will help me find a solution.
My devices : Raspberry pi 3B+ (Jeedom installed, Abeille plugin working), Pizigate (firmware 3.1a), some Xiaomi door sensor, a Xiaomi smoke detector.
 
For a few weeks, everything used to work well, I managed to have a fully working perimeter alarm.
But a few days ago, 2 door sensors and the smoke detector lost the connexion with the pizigate (they were frozen in jeedom) -surely because they are too far from the pizigate. I tried to delete them in Abeille and re-include them, but it was impossible.
The other detectors were working well, and I still wasn't able to include my 3 detectors. So I decided to start from cratch, delete every detectors, reinstall jeedom, reinstall the pizigate and Abeille : but now I can't include any detector (even the detectors that worked well).
I tried to flash the last firmware (3.1a), I used zigate plugin in Jeedom, to try the inclusion : still nothing. I tried a lot of command to erase the pizigate memory (like erasePDM, reset...): still nothing.
I have tested the pizigate with the soft by fairecasoitmeme, and everything seems to be OK.
 
Does anyone have the same problem?
Is there a command to erase every data in pizigate and start from scratch?
 
Thanks a lot !

pipiche38, 2019-10-17:
I would recommend you to contact the Jeedom fox and especially @kiwihc for the Abeille plugin as this looks much more plugin than hardware issue.
As regards to the 'erase every data in pizigate' there is a command to do that 'Erase PDM', but I would also suggest to do it from the Jeedom/Abeille side in order to make sure that you have consistency.

biscwii, 2019-10-19:
Since my last message, I've come to have doubts on the working of the PiZigate, and not the plugins (Abeille or Zigate in Jeedom). Even if this test (https://zigate.fr/documentation/tester-sa-pizigate/) is working well, I still wasn't able to include any detector. I ordered a Zigate-USB, and I managed to include every detector easily. So, the PiZigate seems to have a problem. But I still don't now what problem.

pipiche38, 2019-10-19:
Did you try a software reset of the PiZigate which not intrusive and could a good way to unblock the situation ?

Last did you try an Erase PDM ? This will erase everything in your pizigate and from my experience unblock the situation

biscwii, 2019-10-20:
I have used every reset I could (erase PDM of course, factory reset) but none of these have worked

pipiche38, 2019-11-28:
I would consider a different problem, and might relate to an Hardware issue. pleas ereport to zigate.fr

closed 2020-10-07 by fairecasoimeme