fairecasoimeme/ZiGate #248: 0x0100 - Number of Attributes limit

pipiche38, 2019-10-13:
In regards to #114, is there any limitation from the Zigbee norm on the number of atrributes to be read at once ?

The question is on purpose as I found that Legrand do not answer if more than 4 attributes are requested, and that is the same for Livolo switches.

Basically the behaviour is:

- 0x0100 with 5 attributes no answers from those devices
- 0x0100 with 4 attributes answers from those devices

So the question is , is that brand (implementation) specifics, or by the Zigbee standar ?

closed 2019-11-27 by pipiche38

tcharp38, 2021-05-20:
Hello
I've just found an issue in that way.

I'm interrogating EQ for supported attributes.

[2021-05-20 13:32:53] zgWrite2(010000343702C1D601010102000000001400000001000200030004000500060007000800090010001100120013001400150016001700180019)

So asking for attr 0000 to 0009 on cluster 0102

But I see this on the network :(

[image]

It sounds that there is already an issue when SENDING the request. So EQ may not reply properly.

Any clues ?
As stated by @pipiche38 is there a real limitation we should now on number of attrib interrogated ?
PS: Still zigate v1 FW 31d

Thanks

tcharp38, 2021-05-20:
Note: for info I get answer from EQ as follows which seems completely in line with the wrong request:

[image]

pipiche38, 2021-05-20:
why are you not using 0x0140 to get the list of supported Attributes ?

I think ZiGate is indeed as a limit in terms of number of attributes to be  send in one Read Attribute request

tcharp38, 2021-05-20:
Did you test the 0140 ?
On my side I could not get the reality of supported attributs. I don't think this time it is a zigate/NXP stack limit but just device which do not properly support this discovery mechanism.
If you tell me it works perfectly then I'll recheck. I probably made some mistakes.

In term of limit... what is it ?

pipiche38, 2021-05-20:
Yes, it works fine, then it is up to the remote device to respond properly.

0x0140 EpZiGate EpDest Cluster 0x0000 0x00 0x00 0x0000 0xff

tcharp38, 2021-05-21:
Ok but "up to the remote device to respond properly".
As far as you know is that really supported by all devices ? If not this command is useless for "discovery".

pipiche38, 2021-05-21:
Most of the devices I have seen, respond properly to it

tcharp38, 2021-05-21:
Thanks. I'm gonna check this again and see if reliable. If at least 1 EQ is not responding, I could not (and no one)  rely on it for discovery.

What about attrib limit ? We must give a value. At least one that is garantied to work.

pipiche38, 2021-05-21:
for the Read Attribute, what I see in the code, is a maximum of 10 Attribute. This is limit is given by the firmware (not the NXP stack).
The other consequence to have a large number of attribute is when the device will respond to it, it has to keep the same way and respond to all attribute in once, which mean that the size of the message will be quiet large ( in terms of Zigbee messages)

tcharp38, 2021-05-21:
I m expecting the same size required for 0140 raw answer since if device is supporting 20 attrib it should report them thru 0140 isn't it ?

pipiche38, 2021-05-21:
no, as you'll get 1 8140 per attribute, this is why in the response there is a "Completion/Final" flag

tcharp38, 2021-05-21:
Yes but in hybrid mode there is 8002 too which will contain all supported attrtib. Then message biggest size. Correct ?

tcharp38, 2021-05-21:
Thanks @pipiche38 
Using 0140 + 8002 is ok to get a list of supported attributs.