fairecasoimeme/ZiGate #250: Support  SONOFF BASICZBR3

M4t3o, 2019-10-16:
Hi 

just for informations, i've just seen that sonoff distributes a zigbee component :
https://www.itead.cc/sonoff-basiczbr3.html

I'd like to try this one with my zigate when it could be possible.
I don't find it here : https://zigate.fr/le-materiel-compatible-zigate/

Have u plan to support it ?

Regards.

pipiche38, 2019-10-17:
SONOFF devices are usually well compatible with Zigate, I wouldn't see any reason why it wouldn't be the case here.
They mention that it works with SmartThings hub, so that is quiet positive.

I don't know why application and plugin your are using, but if you are based on Domoticz and the Zigate plugin, I would be more than happy to help you in case of trouble.

pdecat, 2019-10-17:
Isn't the BASICZBR3 the first SONOFF device using the Zigbee protocol?
AFAICT, that's the only one listed on [their Zigbee website](https://sonoff.tech/product/zigbee-diy-smart-switches/basiczbr3).

pipiche38, 2019-10-17:
that is indeed this one

fairecasoimeme, 2019-10-18:
I will order and I'll tell you ;)
Fred

pipiche38, 2019-10-24:
As mentioned in my early comment, indeed it is a supported one and works well with Zigate (and the Domoticz plugin)

Hedda, 2020-04-16:
FYI, SONOFF BASICZBR3 is based on Texas Instruments CC2530 with Z-Stack Home 1.2 firmware with switch/router config and should use the Zigbee Home Automation (HA) application profile specification. For some more information see https://github.com/Koenkk/Z-Stack-firmware/issues/145 and https://github.com/Koenkk/Z-Stack-firmware/issues/153 as well as https://notenoughtech.com/home-automation/sonoff-basic-r3-zigbee/

pipiche38, 2020-06-25:
It is supported ... 
https://zigate.fr/produits-sonoff-compatible-zigate/

tobe closed

closed 2020-06-25 by fairecasoimeme