fairecasoimeme/ZiGate #253: Impossible d'appareiller mes capteurs Xiaomi (nouvelle installation)

sirpapy, 2019-10-26:
Bonjour,

Je viens de recevoir ma clé Zigate USB et je l'ai mis à jour au Firmware 3a.1

J'ai un HP Proliant dans lequel j'ai installé un Jeedom en Docker avec le plugin Zigate.

La clé est bien reconnu par Jeedom (je vois les informations de version etc). Mais quand je passe en mode inclusion pour ajouter un nouveau Device Xiaomi il y a que la lumière bleu qui clignote et après plus rien ne se passe. J'ai beau refaire l'opération(reset Capteur, repassage en mode inclusion) mais j'ai aucun ajout.

 

Pouvez vous m'aidez svp?

 

Merci beaucoup

pipiche38, 2019-10-26:
Vous devriez poster votre question sur le forum ou le github lié à Jeedom et son plugin pour la Zigate.

sirpapy, 2019-10-26:
J'ai posté cela labas mais sans réponse de leur part malheureusement :/

pipiche38, 2019-11-28:
Ne relève pas d'un problèmatique Zigate.
To be closed

pipiche38, 2020-05-08:
Should be close

closed 2020-10-07 by fairecasoimeme