fairecasoimeme/ZiGate #254: Fréquence de raffraichissement trop longue.

sifus, 2019-10-26:
Hello,

Il est 22h47 et j'ai encore les températures de 19h53 su mon Jeedom. De plus l'heure affichée semble être H-1.

Comparativement, mes sondes EnOcean se mettent à jour toutes les 20 minutes.

Jeedom 3.3.34

clé et plugin Zigate :
Version plugin 1.3.5
Version lib 0.32.2
Version firmware 3.1a

Sonde temp Xiaomi

j'ai acheté la clé Zigate et donc je dépend pour certain appareil de cette clé.

doudz, 2019-10-26:
Le décalage d'heure vient d'un bug concernant le fuseau horaire rien de grave

doudz, 2019-10-26:
Et sinon si la température n'a pas changé c'est peut être qu'elle n'a pas changé tout simplement ?

sifus, 2019-10-26:
Merci pour ta réponse.

Ok pour le bug du fuseau.

Par contre pour la température tu te doutes bien que j'ai vérifié.
Tous mes capteurs de température Xiaomi (Zigbee) sont doublé par des capteurs Tryosys (enocean) et la température de ces dernière bouge toutes les 20 minutes même écart minime.

Ce matin j'ai couché ma fille après avoir ouvert les fenêtres pendant 3 heure, il faisait 18C dans la chambre et le radiateur était froid
J'ai regardé mon Jeedom et mon thermostat qui surveillait mon capteur xiaomi n'a pas déclenché le radiateur car le temperature qui était affichée était de 22 degré (température qui datait d'il y a 3h)

doudz, 2019-10-26:
Il faudrait essayer la version bêta du plugin jeedom, qui nettement plus à jour

pipiche38, 2019-10-26:
@sifus 
Simple question, pourquoi poser cette question sur ce github qui est lié au firmware de la Zigate ?

sifus, 2019-10-26:
Parce qu'il ne semble pas y avoir d'autres moyens de contacter la personne à qui j'ai acheté cette clé. Pas de numéro ou d'email pour des questions de SAV.

Et comme le taux de rafraichissement est beaucoup plus fréquentent avec l'app Xiaomi,  je me demande si c'est pas le plugin zigate qui est en cause, ou alors la clé....

pipiche38, 2019-11-27:
Peut etre clos.

sifus, 2019-11-27:
Effectivement c'est un peu mieux depuis les dernière update du plugin.

closed 2019-11-27 by sifus