fairecasoimeme/ZiGate #256: Konke ne s'associe pas

sifus, 2019-11-03:
Hello,

j'ai deux capteurs de mouvement Konke et sur Jeedom avec le plugin Zigate la procédure d'association ne marche pas.

En mode écoute et 10 secondes sur le bouton., mais rien.

Zigate 3.1a
Jeedom 3.3.35

pipiche38, 2019-11-03:
Bonsoir, voir avec les développeurs du plugin jeedom, car cela fonctionne coté Domotciz, mais le plugin a du etre adapté.

sifus, 2019-11-04:
Et du coup c'est qui le développeur Jeedom ?

doudz, 2019-11-04:
@sifus c'est moi, pour tout problème il faudrait ouvrir un ticket ici https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues

doudz, 2019-11-04:
@pipiche38 quel genre d'adaptation ? ce n'est pas une association classique ?

pipiche38, 2019-11-04:
Le Motion Sensor Konke utilise le cluster IAS, il faut donc faire l'enrollment de la Zigate dans le Sensor lors de l'appairage.
Le statut arrive via 0x8401

Il ne faut pas faire de bind du cluster 0x0500 ni de configurereporting

doudz, 2019-11-04:
@pipiche38 merci
donc si je comprend bien si on ne fait pas l'enrollment le sensor est bien appairé mais le statut ne remonte pas c'est ça ?

@sifus merci de confirmer si le problème concerne l'association ou la remontée d'information lors du détection

pipiche38, 2019-11-04:
> @pipiche38 merci
> donc si je comprend bien si on ne fait pas l'enrollment le sensor est bien appairé mais le statut ne remonte pas c'est ça ?

Oui je pense

pipiche38, 2019-11-04:
et ma compréhension est que c'est valble pour chaque device IAS respectant la norme.

doudz, 2019-11-04:
Merci @pipiche38

sifus, 2019-11-04:
Probleme d'association, je n'arrive pas à faire apparaitre le module dans Jeedom.

doudz, 2019-11-04:
Il faudrait relancer le démon en mode debug, retenter de faire l'association et m'envoyer le log

sifus, 2019-11-04:
Ok je te fais ça...

...et d’ailleurs je viens de voir que j'ai souvent un problème de "retry connextion".
Dois-je changer le module USB (j'ai le bleu actuellement) ?

[image]

sifus, 2019-11-04:
En relançant le demon j'ai comme d'habitude tout mes capteurs qui demandent un rafraichissement, ce que je ne fait jamais car je comprend pas comment faire donc j’attends en général que les premières valeurs arrivent toutes seules... (en cliquant sur le bouton rafraichir j'ai jamais l'impression que ça fait quelques chose...)

[image]

doudz, 2019-11-04:
> Ok je te fais ça...
> 
> ...et d’ailleurs je viens de voir que j'ai souvent un problème de "retry connextion".
> Dois-je changer le module USB (j'ai le bleu actuellement) ?
> 
> [image]

est-ce que le réglage du port est sur auto ?

sifus, 2019-11-04:
Et voila mes logs (avec à la fin ma tentative d'association ou rien n'apparait sur jeedom)

`[2019-11-04 12:28:58][DEBUG][requests.packages.urllib3.connectionpool](Thread-1) : Starting new HTTP connection (1): 127.0.0.1
[2019-11-04 13:28:58][DEBUG] : Array (     [action] => ZIGATE_DEVICE_UPDATED     [device] => Array         (             [info] => Array                 (                     [id] => 11                     [mac_capability] => 10000000                     [last_seen] => 2019-11-04 11:58:26                     [ieee] => 00158d00031b3c2b                     [addr] => f94b                     [lqi] => 91                     [power_type] => 0                 )              [generictype] =>              [discovery] =>              [addr] => f94b             [endpoints] => Array                 (                     [0] => Array                         (                             [profile] => 0                             [out_clusters] => Array                                 (                                 )                              [clusters] => Array                                 (                                     [0] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [data] => 0121f90b03281b0421a84305210b00062401000000000a2100000821041064200296230000000000                                                             [attribute] => 65281                                                             [value] => 3.065                                                             [name] => battery_voltage                                                             [type] => float                                                             [unit] => V                                                         )                                                      [1] => Array                                                         (                                                             [value] => lumi.sensor_smoke                                                             [name] => type                                                             [type] => str                                                             [data] => lumi.sensor_smoke                                                             [attribute] => 5                                                         )                                                  )                                              [cluster] => 0                                         )                                      [1] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [data] => 0000000000000000                                                             [attribute] => 255                                                             [name] => zone_status                                                             [value] => Array                                                                 (                                                                     [alarm2] =>                                                                      [supervision] =>                                                                      [alarm1] =>                                                                      [restore] =>                                                                      [trouble] =>                                                                      [low_battery] =>                                                                      [test_mode] =>                                                                      [tamper] =>                                                                      [battery_defect] =>                                                                      [ac_fault] =>                                                                  )                                                          )                                                  )                                              [cluster] => 1280                                         )                                  )                              [in_clusters] => Array                                 (                                 )                              [device] => 0                             [endpoint] => 1                         )                  )          )  )
[2019-11-04 13:28:58][DEBUG] : Recherche eqLogic 00158d00031b3c2b
[2019-11-04 13:28:58][DEBUG] : createcommands for 00158d00031b3c2b
[2019-11-04 13:28:58][DEBUG] : create command 1.0.65281
[2019-11-04 13:28:58][DEBUG] : create command 1.0.5
[2019-11-04 13:28:58][DEBUG] : create command 1.1280.255.alarm2
[2019-11-04 13:28:58][DEBUG] : create command 1.1280.255.supervision
[2019-11-04 13:28:58][DEBUG] : create command 1.1280.255.alarm1
[2019-11-04 13:28:58][DEBUG] : create command 1.1280.255.restore
[2019-11-04 13:28:58][DEBUG] : create command 1.1280.255.trouble
[... cut ...]
[2019-11-04 12:33:57][DEBUG][zigate](ZiGate-Decode data) : Received response 0x8010: b'0003031a'
[2019-11-04 12:33:57][DEBUG][zigate](ZiGate-Decode data) : Dispatch ZIGATE_RESPONSE_RECEIVED
[2019-11-04 12:33:57][DEBUG][zigate](ZiGate-Decode data) : RESPONSE 0x8010 - Version list : major:3, installer:31a, lqi:0, version:3.1a
[2019-11-04 12:33:57][DEBUG][zigate](ZiGate-Decode data) : Dispatch ZIGATE_RESPONSE_RECEIVED
[2019-11-04 12:33:57][DEBUG][zigate](Thread-3) : STATUS code to command 0x0010:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'', lqi:0

`

doudz, 2019-11-04:
merci
Effectivement il ne se passe rien, je ne vois pas de "Device Announce"
Pour le moment le plugin Jeedom est donc hors de cause

Si vous avez bien respecté la procédure https://zigate.fr/produits-konke-compatibles-zigate/compatible/konketempsensor/ il faudrait voir avec @fairecasoimeme 
c'est peut-être un souci de compatibilité avec un modèle en particulier ou un canal ou autre...

sifus, 2019-11-04:
Non je le force en sélectionnant "Prolific..." sinon souvent il se remet sur le port de ma clé enocean (UBS FTDI) et du coup tous mes modules passent en inconnu.

D’ailleurs au début je pensais que c’était un bug avant que je découvre que c’était repassé sur un autre port après un redémarrage ou une coupure de courant.

doudz, 2019-11-04:
> Non je le force en sélectionnant "Prolific..." sinon souvent il se remet sur le port de ma clé enocean (UBS FTDI) et du coup tous mes modules passent en inconnu.
>
> D’ailleurs au début je pensais que [...]

Le changement de port est vraiment un problème, surtout quand on a plusieurs clés similaire.
Comment est reconnu la clé enocean ? (un petit screenshot)

sifus, 2019-11-04:
> c'est peut-être un souci de compatibilité avec un modèle en particulier ou un canal ou autre...

Un gars sur un autre topic dit qu'il a réussi à faire l'association de ces modules (Konke).

Pour la clé enocean, ce screenshot ?
[image]

pipiche38, 2019-11-04:
Konke --> Canal 15,20, 25
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Compatible-Devices.md#konke

doudz, 2019-11-04:
> > c'est peut-être un souci de compatibilité avec un modèle en particulier ou un canal ou autre...
> 
> Un gars sur un autre topic dit qu'il a réussi à faire l'association de ces modules (Konke).
> 
> Pour la clé enocean, ce screenshot ?
> [image]

merci, je voulais simplement voir comment était reconnu la clé car je projette de modifier le mode auto pour la zigate

sifus, 2019-11-04:
> Il faudrait relancer le démon en mode debug, retenter de faire l'association et m'envoyer le log

Voila mes log Zigate Debug (association a la fin)

`[2019-11-04 12:29:03][DEBUG][zigate](Thread-2) : Check Need discovery LUMI lumi.sensor_ht (2e7c) 00158d0002f354d7
[2019-11-04 12:29:03][DEBUG][zigate](Thread-2) : Check Need discovery Device  (096c) 00158d0003269677
[2019-11-04 12:29:03][DEBUG][zigate](Thread-2) : REQUEST : 0x0100 b'\x02	l\x01\x01\x00\x00\x00\x00\x00\x00\x02\x00\x04\x00\x05'
[2019-11-04 12:29:03][DEBUG][zigate](Thread-2) : Msg to send b'010000107502096c01010000000000000200040005'
[2019-11-04 12:29:03][DEBUG][zigate](Thread-2) : Encoded Msg to send b'010211021002101075021202196c021102110210021002100210021002100212021002140210021503'
[2019-11-04 12:29:03][DEBUG][zigate](Thread-2) : Waiting for status message for command 0x0100
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Listen) : Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x86\x02\x10\x02\x12\x02\x11\x02\x10\x02\x10\x03\x01\x87\x02\x12\x02\x10\x02\x18\xe1\xd4\x02\x11\x02\x11\x02\x12\x02\x19l\xa9v\x03'
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Event Loop) : Dispatch ZIGATE_PACKET_RECEIVED
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Decode data) : Received response 0x8000: b'00020100'
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Event Loop) : Dispatch ZIGATE_PACKET_RECEIVED
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Decode data) : RESPONSE 0x8000 - Status response : status:0, sequence:2, packet_type:256, error:b'', lqi:0
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Decode data) : Received response 0x8702: b'd4010102096ca9'
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Decode data) : Dispatch ZIGATE_RESPONSE_RECEIVED
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:212, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:096c, sequence:169, lqi:118
[2019-11-04 12:29:03][DEBUG][zigate](Thread-2) : STATUS code to command 0x0100:RESPONSE 0x8000 - Status response : status:0, sequence:2, packet_type:256, error:b'', lqi:0
[2019-11-04 12:29:03][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:212, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:096c, sequence:169, lqi:118
[2019-11-04 12:29:03][DEBUG][zigate](ZiGate-Decode data) : Dispatch ZIGATE_RESPONSE_RECEIVED
[2019-11-04 12:29:03][DEBUG][root](Thread-1) : Send to jeedom :  {'action': 'ZIGATE_DEVICE_NEED_DISCOVERY', 'device': Device  (a79a) 00158d0002e272b7}
[2019-11-04 12:29:03][DEBUG][requests.packages.urllib3.connectionpool](Thread-1) : Starting new HTTP connection (1): 127.0.0.1
[2019-11-04 13:29:03][DEBUG] : Array (     [action] => ZIGATE_DEVICE_NEED_DISCOVERY     [device] => Array         (             [info] => Array                 (                     [id] => 4                     [mac_capability] => 10000000                     [last_seen] => 2019-11-01 17:39:07                     [ieee] => 00158d0002e272b7                     [addr] => a79a                     [lqi] => 66                     [power_type] => 0                 )              [generictype] =>              [discovery] =>              [addr] => a79a             [endpoints] => Array                 (                     [0] => Array                         (                             [profile] => 0                             [out_clusters] => Array                                 (                                 )                              [clusters] => Array                                 (                                     [0] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [data] => 100021c70b21a843240100000000217200205d                                                             [attribute] => 65282                                                             [value] => 3.015                                                             [name] => battery_voltage                                                             [type] => float                                                             [unit] => V                                                         )                                                  )                                              [cluster] => 0                                         )                                      [1] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [value] =>                                                              [name] => onoff                                                             [type] => bool                                                             [data] =>                                                              [attribute] => 0                                                         )                                                  )                                              [cluster] => 6     [... cut ...]
[2019-11-04 12:58:57][DEBUG][zigate](ZiGate-Decode data) : Received response 0x8010: b'0003031a'
[2019-11-04 12:58:57][DEBUG][zigate](ZiGate-Decode data) : Dispatch ZIGATE_RESPONSE_RECEIVED
[2019-11-04 12:58:57][DEBUG][zigate](ZiGate-Decode data) : RESPONSE 0x8010 - Version list : major:3, installer:31a, lqi:0, version:3.1a
[2019-11-04 12:58:57][DEBUG][zigate](Thread-10) : STATUS code to command 0x0010:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'', lqi:0
[2019-11-04 12:58:57][DEBUG][zigate](ZiGate-Decode data) : Dispatch ZIGATE_RESPONSE_RECEIVED`

doudz, 2019-11-04:
@sifus est-ce que la zigate a été mise sur l'un des canaux compatibles (@pipiche38 merci) 15, 20 ou 25 ?

sifus, 2019-11-04:
> @sifus est-ce que la zigate a été mise sur l'un des canaux compatibles (@pipiche38 merci) 15, 20 ou 25 ?

Ah, c'est ça qu'il voulait dire, non je ne suis jamais intervenu sur les canaux... ou est-ce que je change les canaux ?

là, à la place de canal 11 (ça semble plutôt être un paramètre wifi ...) ?
[image]

doudz, 2019-11-04:
c'est effectivement similaire à ce qu'on trouve pour le wifi
il faudrait mettre 15 par exemple.
Par contre attention, peut-être (je n'ai jamais essayé) il faudra peut-être refaire l'appairage des autres équipements

sifus, 2019-11-04:
Bien joué !
C'est bon j'ai détecté mes capteurs de mouvement Konke en changeant le canal pour le 15 (à la place du 11 par défaut).$

Les autres capteurs demandent effectivement un rafraichissement en changeant de canal.

Du coup est ce qu'il y a un canal mieux qu'un autre selon l'encombrement ou autre (comme le wifi) ?

doudz, 2019-11-04:
D'abord, il faut choisir un canal compatible avec son matériel, et ensuite oui c'est une question d'encombrement, donc choisir un canal différent de son voisin éventuellement peut aider
vérifier aussi les interférences avec le wifi car les fréquences sont similaires

https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

sifus, 2019-11-04:
Ok donc du coup il faut choisir le canal qui fonctionne avec tous mes équipements, du coup le fait que les Konke ne marchent que sur 3 canaux réduit les possibilités de cohabitation ?...

Autre question, il est dit sur cet article qu'il faut corriger un problème de maillage. Mais il n'explique pas comment faire... une idée ?

> Cependant, comme ce sont des capteurs de  » sécurité « , ils passent par un phase d' »enrollement ». Le capteur et le coordinateur doivent passer par un phase d’échange pour bien s’identifier (une sorte de signature).

> Et c’est là la subtilité. Au lieu de s’adresser au endpoint 0x01 (normalement le coordinateur), il envoie sa requête à 0x15. Du coup, les coordinateurs classiques (hormis j’imagine le coordinateur de la marque) ne reçoivent pas la trame et le capteur se met en défaut.

> Une fois ce problème résolu, le capteur de mouvement enverra à chaque détection une trame au coordinateur

http://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/

sifus, 2019-11-04:
Désoler je sais pas si c'est à moi de clore le sujet...

pipiche38, 2019-11-04:
Oui tu peux clore, je pense

closed 2019-11-06 by sifus