fairecasoimeme/ZiGate #257: Perte connexion Capteur porte/fenêtre Xiaomi

xAsKeYx, 2019-11-04:
Bonjour,

J'ai souvent des pertes de connexion sur mes capteurs Xiaomi, je suis obliger de les associer de nouveau pour qu'ils soient fonctionnels

Avez une idée de ce qu'il se passe ?

Pour info : 
Version plugin: 4.6.008
Version Firmware: 031a

Cordialement,

Adorem, 2019-11-04:
En lisant cet article aujourd'hui :
https://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/

J'ai retenu cette partie :
> Chez Xiaomi, lorsque le capteur décide de sortir du réseau suite à des problèmes de communication, il est incapable de rejoindre le réseau. (Je ferai surement un article plus complet sur le sujet). Les capteurs Konke n’ont pas ce problème. Ils réintègrent le réseau dès que la communication est rétablie.

Donc peut-être que @fairecasoimeme peu t'en dire plus ?

xAsKeYx, 2019-11-04:
Merci de ton retour, j'ai oublié de spécifier que j'ai 2 répétiteurs ikea afin de palier normalement à ce problème.

Mais oui je vais voir si ils ont plus d'infos.

xAsKeYx, 2019-11-05:
Je viens de passer sur le plugin 4.6.016, je vais voir si il y a du changement

pipiche38, 2019-11-05:
Il faut aussi savoir que les Xiaomi respectent la norme FCC alors que Ikea, Philipps la norme CE.
Le plugin configure la Zigate en mode CE (pour l'Europe) est du coup réduit la puissance d'émission de la Zigate.

Donc il pourrait etre interessant dans ta configuration de passer la Zigate en mode FCC (qui aura une puissance supérieure pour l'émission, equivalente a celle des Xiaomi). Attention cela implique que la Zigate va etre capable de toucher des devices de type CE, qui eux ne pourront pas communiquer avce la Zigate .

Pour changer ce paramètre, cela se fait via le Web Admin GUI dans les Settings

xAsKeYx, 2019-11-07:
Merci pipiche, j'ai fait la modification hier. Je vais voir si cela corrige mon problème.

xAsKeYx, 2019-11-18:
Bon le problème persiste toujours même en mode FCC, j'ai perdu 3 capteurs depuis la modification.

fairecasoimeme, 2019-11-18:
Bonjour,

Bonjour,

En effet, les xiaomi souffrent d'un problème de "leave rejoin" et de routage. Il semblerait qu'une fois le capteur accroché à une route, il ne change plus.

Il serait intéressant de suivre la procédure suivante.
D'abord appairer les routeurs IKEA. Puis ensuite appairer les capteurs Xiaomi.
Si vous êtes sur de vous, qu'un capteur doit passer par un routeur. Il faut appairer le capteur à 5cm du routeur IKEA.

Combien de capteur avez-vous en tout ?

Fred

xAsKeYx, 2019-11-18:
J'ai 13 capteurs en tout.

J'ai déja fait cela, mais le problème c'est les changement de routeur se font en automatiquement

Topologie du 08/11
[image]
Topologie du 15/11
[image]

Prenons l'exemple du capteur portail, qui au 08/11 était bien sur le répétiteur ikea salle à manger, puis on voit que le 15/11 il est passé en direct sur la zigate. Depuis hier le 17/11 le capteur n'est plus disponible.

En gros il n'aurais pas du changé de routeur puisque le plus prêt étant le rep ikea salle à manger.
Dans les autres cas mes capteurs sont assez proche de la zigate et non pas de problème de portés (sauf celui du chalet)

Cela devient pénalisant  pour ma surveillance car je suis en train de monté des sirènes d'alarme.

pipiche38, 2019-11-18:
Le changement de router est inhérent au fonctionnement de la stack Zigbee, et ni la Zigate ni le plugin n'ont la main dessus.

pipiche38, 2019-11-27:
Ce que je peu confirmer, c'est la reproduction de ce problème. Il semble par contre que cela se produise sur certain Capteurs Xaiomi et pas d'autress.
Typiquement j'ai 7 Temp/Humi/Baro et ce sont toujours les 2 memes qui disparaissent du reseau au boutd 'un certain temps !!

xAsKeYx, 2019-11-27:
oui se sont majoritairement les mêmes aussi de mon côté, par contre j'ai eu 1 seul capteur de température qui a déco une fois (sur 3) et depuis pas de problème

sifus, 2019-12-13:
Pareil de mon coté, j'ai des capteurs qui ne relève plus la température ou la statut d'ouverture d'un coup et assez souvent.
Je suis obliger les re-associée et mème en faisant ça parfois ça ne fonctionne pas pendant plusieurs jours.

doudz, 2019-12-13:
@pipiche38 Est ce qu'en passant en FCC le problème disparaît ?

sifus, 2019-12-13:
C'est quoi la procédure à saisir dans l'interface admin ?

Et j'ai pas compris ce que cela impliquait pour les périphérique CE ?...

pipiche38, 2019-12-13:
> 
> 
> @pipiche38 Est ce qu'en passant en FCC le problème disparaît ?

Non, je ne pense pas que ce soit lié à la qualité de réception. Les capteurs loin de la Zigate sont stables, d'autres proche de la Zigate sont stables également. 
D'autres utilisateurs ont le même problème et a date, pas de point de corrélation, hormis Xiaomi !

Donc pour moi on est sur la une problématique équivalente à celle qui existait il y a un ou deux ans, mais qui semble plus limitée

pipiche38, 2019-12-13:
> 
> 
> C'est quoi la procédure à saisir dans l'interface admin ?
> 
> Et j'ai pas compris ce que cela impliquait pour les périphérique CE ?...

CE versus FCC est le niveau de certification que la Zigate va utilisée. La conséquence, c'est une plus forte puissance d'émission en FCC qu'en CE.
Les devices Xiaomi sont en FCC alors que beaucoup de devices Européen sont en CE ( Ikea, Philips ...). 

Donc le risque quand on est pas dans le bon niveau , c'est que la communication ne fonctionne que dans un sens.

Par exemple en Mode CE, un Device Xiaomi peut joindre la Zigate alors que celle-ci peut ne pas communiquer avec le device Xiaomi car hors de portée (du fait du mode CE).

Par contre inversement si la Zigate est en mode FCC, celle-ci peut etre à meme de joindre une ampoule IKEA par exemple, mais en retour l'ampoule Ikea étant en mode CE peut ne pas joindre la Zigate.

En conclusion si tu n'as que des Xiaomi, tu peux passer en mode FCC, si tu as un mix il faut se mettre au moins disant c'est a dire CE.

sifus, 2019-12-13:
Ok mais du coup, qu'est ce qu'on peut faire ?
Parce que mine de rien je me suis pas mal équipé en xiaomi depuis que j'ai la Zigate car c'était marqué comme compatible. Du coup aujourd'hui ma domotique repose sur un peu sur du Enocean et pas mal sur du Xiaomi (pour la température et les ouvertures).

Le fait que les modules ne se réassocissent par après une perte de connexion c'est ultra embêtant pour la fiabilité du système surtout quand l'alarme compte sur eux.

Est-ce que ça veut dire que le "coté Zigate" ne peut rien régler ?

sifus, 2019-12-13:
Sur la ZiGate j'ai du Xiaomi et Konke

sifus, 2019-12-15:
Bon j'ai tout réasoccié manuellement hier et les modules sont encore dans les choux. Il communiquent dans la panel santé, mais les données sur le dashboard de jeedom sont perimées et ne s'actualisent pas.

[image]

[image]

C'est ultra instable, ça marche pendant 1 mois et au moindre truc genre redemarage de rpi, mise a jour de zigate ou jeedom plus rien ne marche :(

sifus, 2019-12-16:
Capteur toujours bloqués et pourtant ils communiquent :

[image]

[image]

zigate 3.1a
plugin 1.4.1
Jeddom 3.39

doudz, 2019-12-16:
Est-ce que je pourrais voir une capture d'écran de l'historique de température sur la même période ?
Question bête, est-ce que la température a changé durant cette période ?
Le panneau santé indique la dernière fois où le capteur a communiqué en général, pas forcement la température, toutes les 50min il reporte le niveau de batterie par exemple

sifus, 2019-12-16:
Ah oui bien, y'a d'autre infos que la température.

voila la capture :

[image]

closed 2020-10-07 by fairecasoimeme