fairecasoimeme/ZiGate #258: Konke associe mais non focntionnel

sifus, 2019-11-06:
Bonjour,

mes modules de détection de mouvement Konke sont inutilisables.

Je le vois dans mon interface Jeedom mais le statut est bloqué sur "alarme 1 déclenchée".

Une piste ici peut-être mais la réponse n'est pas donnée... comment modifier le endpoint? 

> 
> Et c’est là la subtilité. Au lieu de s’adresser au endpoint 0x01 (normalement le coordinateur), il envoie sa requête à 0x15. Du coup, les coordinateurs classiques (hormis j’imagine le coordinateur de la marque) ne reçoivent pas la trame et le capteur se met en défaut.
> 
> Une fois ce problème résolu, le capteur de mouvement enverra à chaque détection une trame au coordinateur

http://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/

pipiche38, 2019-11-06:
> Links:

Je ne pense pas que le konke envoie une trame de fin d'alarme. C'est 
donc à Jeedom ou le plugin de faire le reset.
En tout les cas, c'est implémenté comme ça coté Domoticz

doudz, 2019-11-06:
Effectivement c'est au plugin de faire un reset, qui n'est pas fait actuellement
par contre @sifus merci d'ouvrir un ticket ici https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues
Sinon c'est compliqué de faire le suivi

sifus, 2019-11-06:
> Effectivement c'est au plugin de faire un reset, qui n'est pas fait actuellement
> par contre @sifus merci d'ouvrir un ticket ici https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues
> Sinon c'est compliqué de faire le suivi

Yes, y'a déjà un post dans lequel j'ai ajouté ma graine :
https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues/214

pipiche38, 2019-11-27:
Je pense qu'on peut clore cet issue

sifus, 2019-11-27:
Yes, pour l'instant j'ai crée le scenario de reset à 0 mais effectivement c'est au plugin de faire ça je trouve...

closed 2019-11-27 by sifus

pipiche38, 2019-12-08:
Bonsoir, 

juste pour vous dire que j'ai réussi à utiliser les Konke Switch, Motion, Door et Temp sur le canal 11.

sifus, 2019-12-09:
Hello, c'est pas un des deux canaux compatible... comment avez vous fait ?

pipiche38, 2019-12-09:
Il faut faire l'association des Konke sur le canal 15, puis une fois associée, basculer la Zigate sur le canal 11, le Konke va suivre.

Attention lors de la bascule sur le canal 11, il faut s'assurer de laisser la Zigate en mode Permit To Join.

sifus, 2019-12-15:
Une procédure particulière ou juste je clic sur "mode inclusion" et après je vais dans les paramètres pour changer le canal ?...

doudz, 2019-12-15:
@sifus sur Jeedom on ne peut pas changer le canal sans relancer le démon. 
Pour pouvoir le faire pendant le permit_join (inclusion) il faut envoyer la commande manuellement avec le terminal.
Se reporter à la documentation ZiGate pour connaître la commande et les paramètres à envoyer.

sifus, 2019-12-15:
j'ai regardé la doc, donc je rentre 0x0049 dans le terminal de la zigate sur jeedom et pendant que ça clignote bleu, je sauvegarde le nouveau canal 11..

J'ai pas besoin desaisir ca : 01 02 10 49 02 10 02 14 B0 FF FC FE 02 10 03 ?
Juste ça : 0x0049

doudz, 2019-12-16:
Non 0x0049 c'est pour fait un permit_join, c'est la même chose que le bouton "Inclusion"
Pour changer le canal sur 11 il faut envoyer :
cmd: 0x0021
data: 00000800

sifus, 2019-12-16:
Ok mais du coup cette commande à part changer le canal elle active le permit join ?

doudz, 2019-12-16:
Non elle n'active pas le permit_join, pour cela il faut cliquer sur le bouton "Mode Inclusion"

sifus, 2019-12-16:
Bon bah j'ai perdu toute mes associations !

Quand on clique sur mode inclusion et qu'on ouvre la console on quitte le mode inclusion.

C'est pas assez claire ces instructions :(

doudz, 2019-12-16:
Le mode inclusion continue pendant 30sec (60sec pour les versions récentes du plugin) donc l'ouverture de la console ne change rien du tout.
Si les associations sont perdus, c'est peut-être la faute au changement de canal justement