fairecasoimeme/ZiGate #263: Sources broken by bad comment

tobiaskohlbau, 2019-11-25:
Looks like the latest sources are broken. It's not possible to build from source without fixing the comment introduced by https://github.com/fairecasoimeme/ZiGate/commit/0de2542f666de2227ce1d0d7e71e9b09bb1db39d.

```
 /*case GENERAL_CLUSTER_ID_COLOUR_CONTROL:
				{
					tsCLD_ColourControlCallBackMessage*    psCallBackMessage =  ( tsCLD_ColourControlCallBackMessage* ) psEvent->uMessage.sClusterCustomMessage.pvCustomData;
```

This comment misses an equivalent end of comment. This results in lot's of build errors issuing bad statements in non switch cases and other stuff. I would assume that the original fix was to comment out the case block. Adding a matching `*/` after the break statement fixed it for me.
I assume build cache was for the rescue to keep building for the newer releases. If my fix gets confirmed by @fairecasoimeme I will prepare a PR.

fairecasoimeme, 2019-11-26:
Oups,
Sorry ... i fixed it.
Fred

tobiaskohlbau, 2019-11-27:
Many thanks for the fast response and solving this on master :-)

closed 2019-11-27 by tobiaskohlbau