fairecasoimeme/ZiGate #264: Device Annouced when Zigate is Full !

pipiche38, 2019-11-27:
It seems that when Zigate is reaching its max number of connected Devices, still the Device Annouced is passed to the above layer ( 0x004D ) .

However, If ZIgate reached its maximum, the device announced should not be brought to the above level. As this is some how providing inconsistency information

Device Annouced means that pairing occured, and in such the above layer rely on that information, while due to the limitation, the device doesn't respond to quesries !

deennoo, 2019-11-28:
That what I'm living every new device

pipiche38, 2019-11-28:
For me there is an issue on the Zigate which consider the Max of 40 (direct ) + 30 (via router) == 70 Devices, while the Zigate should consider its Max Max to 40 with 40 Direct and/or 30 Indirect

pipiche38, 2020-07-08:
I think we can close that one. This was an issue on the plugin side.

closed 2020-07-08 by pipiche38