fairecasoimeme/ZiGate #265: Problème Installation des dépendances // Jeedom

gregdel83, 2019-12-05:
Bonjour,

L'inscription sur le forum du site Zigate étant clos, je pose ce problème ici. 

J'ai fait l'acquisition d'une Zigate-USB installée sur la config suivante : NAS Synology DS218+ / Docker / Jeedom / MariaDB.

L'installation Jeedom & Plugin Zigate s'est passée sans encombre, néanmoins impossible d'installer les dépendances du plugin (dont clé non reconnue & demon KO) 

Pour info j'ai vérifié l'état de Python : https://jeedom-zigate.github.io/jeedom-plugin-zigate/fr_FR/index

J'attache les logs de Zigate Update. 

Merci pour votre aide 
[zigate_update.txt](https://github.com/fairecasoimeme/ZiGate/files/3929214/zigate_update.txt)

doudz, 2019-12-06:
Si c'est possible, je conseille de passer à Jeedom v4

gregdel83, 2019-12-06:
Bonjour et merci.

Je suis déjà en V4.

doudz, 2019-12-06:
c'est corrigé en béta

gregdel83, 2019-12-06:
Bonsoir,

Même sentence/erreur en bêta.

doudz, 2019-12-07:
Attention les modifications ne sont disponibles que le lendemain sur le market Jeedom, ça devrait donc fonctionner avec la mise à jour de la beta d'aujourd'hui

gregdel83, 2019-12-07:
Effectivement, et tout fonctionne depuis votre nouvelle release.

Merci pour votre rapidité.

Je profite du topic pour 2 autres questions :

Savez-vous si les ampoules Smart+ OSRAM sont compatibles, impossible de les
détecter en inclusion ?
Les tâches CRON restent en rouge, est-ce un comportement normal ?

doudz, 2019-12-10:
> Savez-vous si les ampoules Smart+ OSRAM sont compatibles, impossible de les détecter en inclusion ? 

Je dirai oui mais il faut se référer à zigate.fr pour connaitre les compatibilités

> Les tâches CRON restent en rouge, est-ce un comportement normal ?

Oui c'est normal, je n'utilise pas les taches CRON de Jeedom

doeurf, 2019-12-14:
Hello,

J'ai exactement le même problème, comment fait-on pour passer en beta ?

Merci d'avance

Jsl-1, 2019-12-15:
Hello @doeurf !
> Si on parle du plugin comment passer sur la version beta ?

Pour avoir accès au plugin beta, d'après la doc il faut être en jeedom beta, voir la doc ici :
https://jeedom.github.io/documentation/howto/fr_FR/passer_en_beta

doeurf, 2019-12-16:
Hello @Jsl-1 merci, j'avais bien passé le core en beta, mais j'avais pas regardé du côté du profil sur le market ;) J'ai bien pu passer en plugin beta et ça fonctionne de nouveau, plus qu'à rafraîchir mes équipements 1 par 1 💃

pipiche38, 2020-01-13:
to be closed

closed 2020-01-16 by fairecasoimeme