fairecasoimeme/ZiGate #270: IASACE problem !!!

minhtan58, 2020-01-04:
Hi. 
I just adding ACE cluster to Zigate:
1.zcl_options.h
       #define CLD_IASACE
       #define IASACE_SERVER
2. modify .zpscfg
[image]
3. Modify in SDK : Add to struct in control_bridge.h, .c
[image]
But Program is crash when i start it
[image]
By the way, When i run in Origin SW from NXP it work well.
Happy new year!!!

pipiche38, 2020-01-04:
Would be interested to know for which device you want to have IASACE enabled ?

minhtan58, 2020-01-06:
thank for reply. I run Zigate sw on JN5169
for control IAS WD, I use button from I-horn sen request to Coodinator. 
I use JN-AN-1201 as advice.

pipiche38, 2020-01-06:
So why not using  cluster  0X0502 instead ? this will do the same and is implemented in Firmware 3.1a

minhtan58, 2020-01-06:
Thanks you! But now i want add IAS ACE to Zigate for more opiton control! 
I don't know, my sw isn't work when i define ACE cluster
Have nice day!!!

pipiche38, 2020-01-06:
As you want. My understanding is that Zigate is under memory pressure and adding one more cluster server could be challenging.

Enjoy

minhtan58, 2020-01-06:
Yes! I thing so. Is there any way to fix that? Thanks

pipiche38, 2020-06-29:
This will be addressed by some of the PR targeted for 31d.

In between if you have commands to be send from Controler to Devices, you can use the 0x0530 (raw mode) command to issuse then. In 31d you'll get upcoming unmanaged clusters via 0x8002

closed 2020-10-07 by fairecasoimeme