fairecasoimeme/ZiGate #274: Réponses 0x8035 E_PDM_SYSTEM_EVENT_LARGEST_RECORD_FULL_SAVE_NO_LONGER_POSSIBLE

pdecat, 2020-01-26:
Bonjour @fairecasoimeme,

depuis que j'ai mis à jour ma Zigate en v3.1b puis v3.1c qui apporte le support des réponses 0x8035, je m'aperçois que j'en reçois de nombreuses avec le code statut `03` :

```
2020-01-26 12:07:33 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x805\x02\x10\x02\x16C\x02\x13\x02\x10\x02\x10\xf1\x02\x12\x02\x10\x03\x01\x805\x02\x10\x02\x16E\x02\x13\x02\x10\x02\x10\xf1\x02\x14\x02\x10\x03\x01\x805\x02\x10\x02\x16@\x02\x13\x02\x10\x02\x10\xf1\x02\x11\x02\x10\x03\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13\x02\x10\x03'
2020-01-26 12:07:33 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f102'
2020-01-26 12:07:33 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-01-26 12:07:33 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f104'
2020-01-26 12:07:33 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x02', lqi:0
2020-01-26 12:07:33 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f101'
2020-01-26 12:07:33 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x04', lqi:0
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-01-26 12:07:33 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x01', lqi:0
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f103'
2020-01-26 12:07:33 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x03', lqi:0
2020-01-26 12:07:33 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
```

Ce code `03` semble correspondre à [E_PDM_SYSTEM_EVENT_LARGEST_RECORD_FULL_SAVE_NO_LONGER_POSSIBLE](https://github.com/fairecasoimeme/ZiGate/blob/v3.1c/Module%20Radio/Firmware/src/sdk/JN-SW-4170/Components/PDM/Include/PDM.h#L176) qui d'après la doc NXP signifie :

> The EEPROM occupancy is such that the largest record in the PDM can no longer be fully saved.
> `u32EventNumber` carries  the `u16IdValue` of the record that was being processed.

https://www.nxp.com/docs/en/user-guide/JN-UG-3116.pdf

La liste des réponses différentes avec ce code `03` n'est pas très longue :
```
root@hassio:/usr/share/hassio/homeassistant# grep "Received response 0x8035" home-assistant.log |cut -d' ' -f 10 | sort | uniq | grep "b'03"
b'0300000001'
b'030000f005'
b'030000f100'
b'030000f101'
b'030000f102'
b'030000f103'
b'030000f104'
```

Note: je filtre sur `b'03` car j'ai aussi des [`b'07` qui semblent a priori moins grave](https://github.com/fairecasoimeme/ZiGate/blob/v3.1c/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_start.c#L1071).

Mais par contre elle reviennent très souvent, ~17000 par jour :

```
root@hassio:/usr/share/hassio/homeassistant# grep "Received response 0x8035" home-assistant.log | grep ^2020-01-26 | grep "b'03" | wc -l
5363
root@hassio:/usr/share/hassio/homeassistant# grep "Received response 0x8035" home-assistant.log | grep ^2020-01-25 | grep "b'03" | wc -l
17007
```

Comment faut-il interpréter ces réponses ?
Certains appareils seraient-ils saturés ?

jBouyoud, 2020-02-12:
Bonjour @fairecasoimeme ,

J'ai le même problème que @pdecat. 
Suite à la mise à jour de ma zigate depuis la version 3.0f.

J'ai également noté : 
```
DEBUG:zigate:Received response 0x8000: b'14000100'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:20, sequence:0, packet_type:256, error:b'', lqi:0
ERROR:zigate:Command 0x0100 failed E_ZCL_ERR_ZBUFFER_FAIL : b''
```

Est la mémoire de la zigate qui est saturé ou des devices ?

pdecat, 2020-02-22:
Pour ma part, je n'ai aucune trace de `E_ZCL_ERR_ZBUFFER_FAIL` dans mes logs.

pdecat, 2020-02-27:
Pour info, j'ai re-flashé ma Zigate en v3.1c et effacé l'EEprom et la flash pour repartir de zéro et je vois toujours les mêmes messages 0x8035 :

```
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f101' LQI:0
home-assistant.log:2020-02-27 08:47:53 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000642030000f10300
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f103' LQI:0
home-assistant.log:2020-02-27 08:47:53 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000642030000f10300
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f103' LQI:0
home-assistant.log:2020-02-27 08:47:53 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000643030000f10200
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f102' LQI:0
home-assistant.log:2020-02-27 08:47:53 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000645030000f10400
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f104' LQI:0
home-assistant.log:2020-02-27 08:47:53 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000642030000f10300
home-assistant.log:2020-02-27 08:47:53 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f103' LQI:0
home-assistant.log:2020-02-27 08:47:53 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:48:09 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000645030000f10400
home-assistant.log:2020-02-27 08:48:09 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f104' LQI:0
home-assistant.log:2020-02-27 08:48:09 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:48:09 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000642030000f10300
home-assistant.log:2020-02-27 08:48:09 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f103' LQI:0
home-assistant.log:2020-02-27 08:48:09 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:49:11 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000645030000f10400
home-assistant.log:2020-02-27 08:49:11 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f104' LQI:0
home-assistant.log:2020-02-27 08:49:11 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
home-assistant.log:2020-02-27 08:49:11 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8035000642030000f10300
home-assistant.log:2020-02-27 08:49:11 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8035 b'030000f103' LQI:0
home-assistant.log:2020-02-27 08:49:11 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
```

Note: au passage je suis en train de passer à zigpy / zigpy-zigate, d'où le nouveau format de logs.

pipiche38, 2020-02-27:
Voir avec home assistant et le plugin concerné.
Effectivement le firmware 3.1c envoie des information supplémentaire lié à l'utilisation de la PDM

pdecat, 2020-02-27:
Ok mais comment les interpréter ?

Y a-t-il d'autres app qui savent le faire ? Par exemple Domoticz ?

pipiche38, 2020-02-27:
> Ok mais comment les interpréter ?
> 
> Y a-t-il d'autres app qui savent le faire ? Par exemple Domoticz ?

Dans Domoticz, le message est simplement géré en interne et ne remonte pas à l'utilisateur ni dans les Logs. Il sera activé si besoin.

pdecat, 2020-02-27:
Ok, mais est-ce que mon interprétation dans la description de l'issue est correcte ?

> Ce code `03` semble correspondre à [...]

>  Certains appareils seraient-ils saturés ?

pipiche38, 2020-05-08:
Should be close not a firmware issue

badzz, 2020-05-08:
TLDR not an issue
It is not a real issue, it is just a warning
When saving to flash it does a copy backup of *where *you are going to
write in the flash (in case of failure), but when the remaining space is
smaller than the largest segment in use, then you can't do this backup.
Hence the warning, but
1- the flash is not full, there is still space to save new things
2- your data will be saved
It is just that you do not have this safety net in case some flash failure
happens.

pdecat, 2020-05-08:
Ok then, thanks for the inputs.

closed 2020-05-08 by pdecat

ruimarinho, 2020-08-28:
@badzz in my case it appears I get this warning from a specific sensor but I am unable to either receive other sensor information or even attempt to repair. It's like if the sensor got blacklisted by zigate.