fairecasoimeme/ZiGate #275: maj pizigate

xav31190, 2020-02-01:
Bonjour,
Je cherche sans succés un tuto pour mettre à jour ma pizigate ( non usb, pluggé sur mon pi3.
puis je le passer en 3.1c ? via le lien ZiGate_Coordinator_v3.0e.bin ?
Merci pour vos réponses

Tobby-NZ, 2020-02-23:
Bonjour,

Je viens tout juste de mettre à jour ma PiZigate sans problème particuliers : 

Mettre en conf Flash la PiZigate : 
https://zigate.fr/documentation/parametrer-la-pizigate/
```bash
gpio mode 0 out
gpio mode 2 out
gpio write 2 0
gpio write 0 0
gpio write 0 1
```

Télécharger la realease souhaitée :
https://github.com/fairecasoimeme/ZiGate/releases
```bash
wget https://github.com/fairecasoimeme/ZiGate/releases/download/v3.1c/ZiGate_v3.1c.bin
```

Stopper toute application utilisant la PiZigate

Mettre à jour la PiZigate : 
https://zigate.fr/documentation/mise-a-jour-de-la-zigate/
```bash
./JennicModuleProgrammer -V 6 -P 115200 -f ZiGate_v3.1c.bin -s /dev/ttyAMA0
```

Remettre en production la PiZigate : 
```bash
gpio mode 0 out
gpio mode 2 out
gpio write 2 1 
gpio write 0 0
gpio write 0 1
```

Tester la PiZigate : 
https://zigate.fr/documentation/tester-sa-pizigate/
```bash
sudo ./PiZiGate_test /dev/ttyAMA0
```

J'espère que tout sera good pour toi !

math8406, 2020-04-09:
Merci.

A ajouter a la doc officielle ?

pipiche38, 2020-04-09:
> Bonjour,
> Je cherche sans succés un tuto pour mettre à jour ma pizigate ( non usb, pluggé sur mon pi3.
> puis je le passer en 3.1c ? via le lien ZiGate_Coordinator_v3.0e.bin ?
> Merci pour vos réponses

Tu devrais trouver ces informations chez les fournisseurs des plugins.
Voici celle pour Domoticz et son plugin Zigate 

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/PiZigate-RPI3B%2B-Cookbook.md

pipiche38, 2020-05-08:
Issue should be close

closed 2020-10-07 by fairecasoimeme