fairecasoimeme/ZiGate #279: Question : Zigate compatibilité Profalux et Xiaomi

jBouyoud, 2020-02-13:
Bonjour @fairecasoimeme , 

J'ai une question à propos de la compatibilité de la Zigate (Firmware 3.1c) et les produits profalux / Xiaomi Aquara. 
De ce que j'ai compris, Profalux et Xiaomi utilisent le protocole Zigbee Home Automation 1.2. (https://zigate.fr/documentation/association-des-volets-profalux-a-la-zigate/).

Il y a apparemment eu un changement dans le firmware zigate  (https://zigate.fr/2019/07/03/note-dinformation-juillet-aout-2019/) . 
Est ce que cela veut dire qu'à partir du Firmware 3.1a, la Zigate n'est plus compatible avec les produits de ces marques là ? 

Si oui, Est ce que cela veut dire que je suis contraint d'utiliser le Firmware alternatif (https://github.com/fairecasoimeme/ZiGate_HA1.2) ?
Si oui, est ce que celui ci est compatible avec les différentes intégrations logiciels (notamment avec https://github.com/doudz/zigate) ? 

Merci pour vos réponses

pipiche38, 2020-02-13:
Comme indique le firmware ZiGate_HA1.2 est pour "ZiGate firmware specially for Pluzzy devices ".

Pour Profalux comme Xiaomi il faut utiliser le firmware Standard de la Zigate qui est maintenant le 3.1c et qui est compatible avec Profalux, Xiaomi et les autres marque ZigBee 3.0

jBouyoud, 2020-02-13:
Bonjour, 

Merci pour cette réponse. 

Je suis désolé mais le `HA1.2` dans le nom du firmware est confusant vis à vis de la phrase : 
> Les volets Profalux de la gamme Zoé communiquent en ZigBee. Dans leur dernière version, ces volets communiquent grâce au protocole ZigBee HA 1.2.

Il est aussi dit plus bas dans l'article : 
> Donc j’ai ouvert un nouveau github ZiGate_HA 1.2 avec un firmware qui pourra être compatible avec certains matériels Pluzzy et d’autres matériels comme Xiaomi (par exemple)

Pour être sûr de bien comprendre la Zigate gère la layer Zigbee mais pas et le Zigbee Home automation (ZHA - qui doit être une layer application au dessus 🤷‍♂ ) ? 

Je pose toutes ces questions car ma Zigate est bien en 3.1c et j'ai quelques problèmes avec mes volets Profalux lorsqu'ils sont pilotés par la Zigate (par exemple : on/off fait remonter le volets plus haut que e réglage en butée; comme si il y avait un overflow) et je pers régulièrement la synchro avec les capteurs Xiaomi. Et j'essai d'en chercher la cause.

pipiche38, 2020-02-13:
Pour ce qui est de Profalux, tu peux regarder par ici: 
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Profalux-corner.md

Profalux ne recommande pas l'usage du On et du Off via le cluster 0x0006, mais plutot via le cluster 0x0008

tcharp38, 2020-03-31:
Salut.
J'allais créer un autre sujet "routage profalux & xiaomi" mais ca fait peut etre aussi du sens ici.
En gros 5 volets profalux et qq capteurs de temp Xiaomi. Les 5 volets couvrent l'appart entier donc je m'attends à un maillage réseau parfait ce qui est loin d'etre le cas.
D'ou qq question de routage
- Qui décide de la meilleure route ? (chaque routeur ? ou le coordinateur ?)
- La route est elle statique (mise en place lors de l'inclusion) ou dynamique ? Si un des volets tombe en rade et ne fait plus son role de routeur, que se passe t'il ?
- Enfin y a t'il une solution pour optimisier le routage ? Les elements ne sont en effet par forcement inclus dans l'ordre optimal.

Qqn saurait répondre à ca ?
Merci

pipiche38, 2020-03-31:
Le routage est géré/décidé par la stack Zigbee , nous n'en avons pas le controle.

Si tu as un router qui est eteint et bien cette route tombe, d'où le principe dun maillage dense, afin de multiplier le nombre de route et de chemins redondants.

tcharp38, 2020-03-31:
Merci pour ce retour rapide.
Donc
- les devices eux meme choisissent leur route preferée (pas d'influence ni du router ni du controlleur)
- Mes xiaomi se fichent de la couverture en router profalux du coup pas d'autre choix que d'avoir la zigate au milieu pour etre aussi proche que possible de chaque capteur Xiaomi
Mais du coup quelle(s) recommandation(s) pour couvrir une surface + large avec 1 seule zigate ?

pipiche38, 2020-03-31:
La recommendation, c'est augmenter le nombre de routers (c'est à dire avoir des Prise Connecté, des Inter avec Neutre, des routers ( Ikea Repeter ... )

PS/ https://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/

et également faire attention au Wifi 2.4 Ghz ( https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Co-existence-of-IEEE-802.15.4-at-2.4-GHz-and-Zigbee.md )

tcharp38, 2020-03-31:
Super. Merci pour ces infos tres interessantes.
Néanmoins, et désolé de revenir sur le sujet, mais tu dis que xiaomi prefere ce connecter au coordinateur en direct et j'ai 5 routeurs (profalux) sur 100m2.
Est ce que tu penses que d'autres types de routeurs fonctionneraient mieux (Ikea, self made..) ?
Qu'est ce que je loupe ?

pipiche38, 2020-03-31:
Voici ce que j'ai observé ... donc relativement empirique

- Les devices s'ils le peuvent se connectent/s'appairent directement à la Zigate . Ce qui ne me choque pas, car autant se connecter en direct sur le contrôleur. En effet c'est cette route qui est le moins de risquée, car le seul risque résiduel est: si le contrôleur tombe, le réseau tombe !

- Ajouter d'autre routers ne ferons que densifier le maillage et ajouter des multiples routes.

tcharp38, 2020-03-31:
Merci pour ton retour. Je vais reflechir à tout ça et continue mes experiences. Les objets Xiaomi sont tellement attractifs en terme de qualité, prix et fonctionnalités que ca vaut le coup de creuser.

ricky074, 2020-03-31:
Salut,
Perso je suis un peu dans le même cas de toi.
J'ai acheté 3 prises ikea tradfri pour densifier le maillage. Après plusieurs jours, voir 1 à 2 semaines, les capteurs xiaomi finissent par passer par les ces routeurs ikea... Mais cela se fait tout seul

tcharp38, 2020-03-31:
Salut @ricky074. Donc ils preferent Ikea que Profalux :)
Etrange comportement. Ou qqch n'est pas respecté chez Profalux.
Quoi qu'il en soit je testerai cette option. Merci pour l'info.

pipiche38, 2020-03-31:
Je ne pense pas que ce soit un problème de préférence ou de respect de norme.
J'ai 7 VR Profalux sur 130M2 , mais j'ai aussi des Ampoules (Ikea, Aqara) , Prise de courant (Legrand, Ikea), Inter w/o Neutre (Legrand), et ce qu'il se passe c'est que les End Devices vont éventuellement voir leur route changer (a la demande de qui. aucune idée, il faudrait voir ce que dit la norme). 

Un peu plus de lecture en rapport avec ce sujet:
https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

tcharp38, 2020-03-31:
Vraiment très interessant ce forum. Merci @pipiche38.
J'en conclus qu'il faut un maillage important mais surtout avec du matos "réputé compatible" Xiaomi.

pipiche38, 2020-05-08:
Should be close

closed 2020-07-24 by jBouyoud