fairecasoimeme/ZiGate #289: Compatibilité XIAOMI MIjia Light sensor

gdseb, 2020-04-28:
Marque | Xiaomi MIjia
Modèle | GZCGQ01LM

L'inclusion du matériel est ok mais la remontée d'information (luminosity) ne fonctionne pas.

Dans les logs du plugin zigate,  j'ai les erreurs suivantes: 
[2020-04-28 13:38:49][INFO][zigate](ZiGate-Decode data) : No template found for lumi.sen_ill.mgl01
[2020-04-28 13:38:49][ERROR][zigate](ZiGate-Decode data) : Command 0x0140 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
[2020-04-28 13:39:12][WARNING][zigate](ZiGate-Decode data) : Unknown response 0x8035

Version plugin 1.6.4
Version lib 0.38.6
Version firmware 3.1c
Port clé ZiGate

fairecasoimeme, 2020-05-07:
Hi,
I think you have to ask the plugin developper. 
Fred

pipiche38, 2020-05-07:
It works on Domoticz, but you need to bind the Cluster in order to get the infor (which is not the case for the previous LUMI devices)

gdseb, 2020-05-08:
OK. Thanks for your reply but i am on Jeedom V4

tcharp38, 2020-05-08:
@gdseb 
This sensor is known to be badly supported by Jeedom but there is some work ongoing.
Anyway, you should use Jeedom issues and not Zigate.
See there: 
https://github.com/KiwiHC16/Abeille/issues

pipiche38, 2020-06-29:
should be close, this is an application issue. Thanks

closed 2020-06-30 by fairecasoimeme