fairecasoimeme/ZiGate #292: Support LEPTITER TRADFRI LED SPOT

Dakey86, 2020-05-07:
Hi,

Does ZIGATE support the SPOT LED LEPTITER from Ikea ?
I use zigate plugin in jeedom, and I'm getting this product ?

If not how can I help you for support this ?

fairecasoimeme, 2020-05-07:
Hi,
I'm think that will be OK. Tell us when you test.
Fred

Dakey86, 2020-05-07:
Ok les commandes remontent bien. Et tout les paramètres sont pilotables. 
Dans Jeedom l'objet est créé donc c'est ok. Il manque l'image d'illustration dans l'interface Jeedom c'est tout.

J'ai une question, pour la jumeler l'ampoule avec un bouton comme celui là 
https://www.ikea.com/fr/fr/p/tradfri-variateur-dintensite-sans-fil-blanc-70408595/
Je dois tout inclure (bouton et ampoule) via Zigate dans Jeedom puis faire la manip via TESTGUY pour mettre l'ampoule dans le groupe de la télécommande. 

Peut on le faire via le terminal du plugin?

pipiche38, 2020-05-08:
Please do check with the plugin developper !
This GitHub is dedicated for the Zigate Firmware

closed 2021-04-20 by fairecasoimeme